microsoft/AirSim #1858: 360 camera view from airsim car

genie0204, 2019-03-29:
I get that there is 5 current cameras that can be activated, 
But am I able to get a 360 camera view from the prospective from the car ? 
Is the only way to get this by getting more camera via ue4?

msb336, 2019-03-29:
@genie0204 there is no 360 camera view in airsim or in the stock unreal engine. To get a full 360 degree view, you will have to stitch individual camera images into a panorama after capturing them from the environment

madratman, 2019-04-02:
@genie0204 also, see discussion here https://github.com/Microsoft/AirSim/issues/1769

csuwp, 2019-12-02:
Have you solved this problem?  I am also want to get the panorama image. DO you have any solution?

botkevin, 2020-11-12:
I have a hackish solution that I posted @ https://github.com/microsoft/AirSim/issues/1769