microsoft/AirSim #1864: How can I add other Vehicle types and features to Airsim?

sinanonur, 2019-04-01:
First of all is it possible to add other vehicle types other than cars and drones (for example boats or submarines,...) by modifying the source code? I am assuming yes since I don't see why not. :)

I have tried reading the documentation and had a look at the source code but It is more time consuming than I expected because I couldn't find related documentation or pointers.

Current documentation is mainly focused on using the Library. For the architecture part I also read the whitepaper. From what I understand from the documentation I will have to:
* Add a new Vehicle Model in Airsim
* Add new Sensor models if not supported
* Perhaps add a new type of controller
* Adjust Physics engine if some fundamental function is not supported (not sure if needed).
* Add wrappers to the necessary environments (Unreal or Unity)

I'd appreciate if I am wrong in any of those. Apart from those:
* How is the flow diagram for an example simulation What calls what and what kind of information is communicated?
* Where can I find the details on what I should Implement an what those are (I presume it should be simillar to what we have in `vehicles/car/api` in airlib
* Where Exactly is the mappings from controls to the applied forces mentioned in the white paper?
* What wrappers are required for both Unreal and Unity environment?

Long story short I want to have a better picture of how the system works and I am not sure where to go on.

These are many questions at once but I'd be happy if information is provided. I may also contribute by adding the other vehicle type that I develop (If sucessfull).

Tnak you.

whxru, 2019-04-02:
Hope this [repo](https://github.com/mitchellspryn/UrdfSim) could be helpful.

mitchellspryn, 2019-04-04:
Author of the other repo here. 

The linked repo is good for robots that can be described in terms of simple kinematics (arms, skid-steer bots, etc). It has some limitations that are relevant to point out: It does not handle floating bots such as boats - that requires a bit of custom setup in the unreal engine. It also does not have unity support.

closed 2019-10-08 by msb336

rachillesf, 2020-05-24:
@mitchellspryn, Is it possible to simulate car-like vehicles with more advanced kinematics such as truck-trailer systems?

mitchellspryn, 2020-05-24:
Possible? Sure. 

What you want to do? Probably not. 

Simulating a car is super complex. A lot of work has gone into UE to make simulation of the car accurate - even down to simulating the movement (slip) of the tires. In the car mode, AirSim uses that class library. UrdfSim does not - its goal is to model simpler robots. You have 2 options (aside from starting from scratch and rolling your own):

1) Dig into the code of AirSim and modify the car model / code to suit your needs. 
2) Use UrdfSim and accept that the simuation isn't going to be very high quality.

plebey, 2023-08-16:
@sinanonur Good afternoon, is there any progress on your development? I am working on my master thesis on the topic of creating a simulation environment for UAVs and would like to use Airsim as a basis. So I need to find a way to add new drones. For my level it seems impossible to understand the project code without any documentation on its architecture. So if suddenly you have something to share - I would appreciate it (and maybe even suggest an alternative to airsim...?) Thanks