microsoft/AirSim #1865: Lidar sensor performs differently in different scenes with same settings

whxru, 2019-04-02:
Hello everyone, I'm trying to make the lidar sensor become a downward obstacle detector within a range, but with the same settings.json, the lidar sensor performs differently in Blocks and CityEnviron which I download directly from the github release (version 1.2.1).

The perfom in Blocks:
[image]

The perfom in CityEnviron:
[image]

The settings.json:
```json
{
	"SettingsVersion": 1.2,
	"SimMode": "Multirotor",
	"Recording": {
		"Cameras": [
			{
				"ImageType": 2,
				"PixelsAsFloat": true,
				"Compressed": false
			}
		]
	},
	"CameraDefaults": {
		"CaptureSettings": [
			{
				"ImageType": 0,
				"Width": 1920,
				"Height": 1080
			},
			{
				"ImageType": 2,
				"Width": 480,
				"Height": 270
			}
		]
	},
	"Vehicles": {
		"COOLA-Drone": {
			"VehicleType": "SimpleFlight",
			"X": 0, "Y": 0, "Z": 0,
			"Sensors": {
				"BottomObstacleDetector": {
					"SensorType": 6,
					"Enabled": true,
					"Range": 5,
					"NumberofChannels": 4,
					"RotationsPerSecond": 10,
					"PointsPerSecond": 1000,
					"X": 0, "Y":0, "Z": 0,
					"Roll": 0, "Pitch": 0, "Yaw": 0,
					"VerticalFOVUpper": -80,
					"VerticalFOVLower": -100,
					"HorizontalFOVStart": 0,
					"HorizontalFOVEnd": 180,
					"DrawDebugPoints": true,
					"DataFrame": "SensorLocalFrame"
				}
			}
		}
	}
}
```

Besides potential bug I wanna ask a question that what exactly the "HorizontalFOVStart" and "HorizontalFOVEnd" mean? Why could I get a complete scanning area like a circle with HorizontalFOVStart 0 and HorizontalFOVEnd 180? It seems like I should get this with HorizontalFOVEnd 359.

msb336, 2019-04-02:
@whxru 

This is definitely a logic issue in airsim, but the reason you are having unexpected LiDAR response is because of your vertical field of view range. your horizontal fov is only a half circle, so the max angle for vertical fov is -90/90 degrees. Going over this value causes a problem with the sensor constructor.

The underlying problem will require a fix inside of airsim, but a workaround for you would be to set your `VerticalFOVLower` value to `-90`

whxru, 2019-04-08:
@msb336 Thanks for your reply, I agree with you that going over the maximum angle might be one of the reasons, but I tried this settings.json:
```json
"Sensors": {
	"BottomObstacleDetector": {
		"SensorType": 6,
		"Enabled": true,
		"Range": 5,
		"NumberofChannels": 4,
		"RotationsPerSecond": 10,
		"PointsPerSecond": 1000,
		"X": 0, "Y":0, "Z": 0.01,
		"Roll": 0, "Pitch": 0, "Yaw": 0,
		"VerticalFOVUpper": -88,
		"VerticalFOVLower": -92,
		"HorizontalFOVStart": 0,
		"HorizontalFOVEnd": 359,
		"DrawDebugPoints": true,
		"DataFrame": "SensorLocalFrame"
	}
}
```
But the performance in CityEnviron:
[image]
the performance in Blocks:
[image]
The perfomance in Blocks is exactly what I want, but I'll run my program in CityEnviron.

whxru, 2019-04-10:
@msb336 And I found that in CityEnviron the point cloud of lidar data is empty after change the direction of drone(by setting yaw argument in `moveByVelocityAsync()` and in Blocks I got an error "AttributeError: 'LidarData' object has no attribute 'pose'" while using `getLidarData()`.

lcyawym, 2019-05-02:
@whxru Hello,I also meet the question same with you.Have you solved this problem?

whxru, 2019-05-03:
@lcyawym Not yet, but I use the default configuration and distinguish those points after getting them.

lcyawym, 2019-05-03:
@whxru  I solved this problem.You can update your airism  or go to the path of the question where remind lost pose and add pose = Pose() . It will work.

msb336, 2019-10-08:
I think your problem may be solved by [this pr](https://github.com/microsoft/AirSim/pull/2143)

ironclownfish, 2019-10-08:
The PR has been approved.

closed 2019-10-08 by ironclownfish