microsoft/AirSim #1866: Airsim compilation error：

a11196690, 2019-04-02:
After compiling successfully using build.cmd in Airsim, I get the following error when running the project file. My UE4 version is 4.18, and I downloaded the latest Airsim.
[image]

madratman, 2019-04-08:
Are you on win 7?

lovettchris, 2019-04-16:
That seems to be Unreal related, and not an AirSim thing... perhaps related to precompiled header?  Does this discussion help?

https://github.com/20tab/UnrealEnginePython/issues/214

Notice that every header file in the AirSim plugin folder includes this Unreal header first:

```
#include "CoreMinimal.h"
```

Perhaps you added a library that does not abide by this rule.  If so perhaps you should build your library as a lib and include the lib instead of trying to build your library as source code in the unreal project.

jonyMarino, 2020-08-14:
Please feel free to open a new issue if you still have this problem, and include the following (as requested when opening a bug issue)

*Make sure to write all reproduction steps
*Include full error message in text form
*Search issues for error message before filing issue
*Attach screenshot if applicable
*Include code to run if applicable

Thanks!

closed 2020-08-14 by jonyMarino