microsoft/AirSim #1869: Cannot record multiple camera images at once in Unity (and problems with Python API)

tjxjb, 2019-04-04:
I have AirSim environments set up for both Unreal and Unity. Everything works as expected in Unreal, but I want to use Unity for the Windridge map. I cannot record multiple images at once in the Unity version. My settings.json is as follows (same settings work fine in Unreal) :

`{
  "SeeDocsAt": "https://github.com/Microsoft/AirSim/blob/master/docs/settings.md",
  "SettingsVersion": 1.2,
  "ViewMode": "",
  "ClockSpeed": 3.0,
  "Recording": {
    "RecordInterval": 0.05,
    "Cameras": [
        { "CameraName": "0", "ImageType": 3, "PixelsAsFloat": false, "Compress": true },
        { "CameraName": "0", "ImageType": 5, "PixelsAsFloat": false, "Compress": true },
        { "CameraName": "0", "ImageType": 0, "PixelsAsFloat": false, "Compress": true }
    ]
  },
  "CameraDefaults": {
    "CaptureSettings": [
      {
        "ImageType": 0,
        "Width": 600,
        "Height": 300,
        "FOV_Degrees": 90,
        "AutoExposureSpeed": 10,
        "AutoExposureBias": 0
      },
      {
        "ImageType": 5,
        "Width": 600,
        "Height": 300

      },
      {
        "ImageType": 3,
        "Width": 600,
        "Height": 300

      }
    ]
  },
  "SubWindows": [
    {"WindowID": 0, "CameraName": "0", "ImageType": 3, "Visible": false},
    {"WindowID": 1, "CameraName": "0", "ImageType": 5, "Visible": false},
    {"WindowID": 2, "CameraName": "0", "ImageType": 0, "Visible": true}    
  ],
  "DefaultSensors": {
        "Barometer": {
             "SensorType": 1,
             "Enabled" : true
        },
        "Gps": {
             "SensorType": 3,
             "Enabled" : true
        },
	"Imu": {
             "SensorType": 2,
             "Enabled" : true
        },
        "Lidar1": { 
             "SensorType": 6,
             "Enabled" : false,
             "NumberOfChannels": 16,
             "PointsPerSecond": 10000
        }
    }
}
`

As a work around, I've tried to use the Python API to get multiple images. Apart from making the simulation run very slowly at high frequency (which is ok, I don't need it to be realtime), I cannot get the recorded images to change resolution even when the settings.json width/height are changed (the images are always just 256x144). Python code is as follows:

`import setup_path 
import airsim
import numpy as np

import cv2
import time

client = airsim.MultirotorClient()
client.confirmConnection()

start = time.time()

while (cv2.waitKey(1) & 0xFF) == 0xFF:
	milliseconds = (time.time() - start) * 1000
	responses = client.simGetImages([
	airsim.ImageRequest(0, airsim.ImageType.Scene, False, False),
	airsim.ImageRequest(0, airsim.ImageType.Segmentation, False, False)
	])
	for i, response in enumerate(responses):
		img1d = np.fromstring(response.image_data_uint8, dtype=np.uint8) 
		img_rgba = img1d.reshape(response.height, response.width, 4)  
		img_rgba = np.flipud(img_rgba)
		airsim.write_png(str(milliseconds) + '_' + str(i) + '.png', img_rgba)
	time.sleep(1)`

Any help would be appreciated!
Thanks!

howard-mahe, 2019-04-10:
Hi,

I did not tried to record images using the "Recording" features but I did with the Python API. I found out the field "CameraDefaults" and "SubWindows" in settings.json were ignored. In order to change the output image dimensions, I directly modified the end of the file [AirSimSettings.cs](https://github.com/Microsoft/AirSim/blob/master/Unity/UnityDemo/Assets/AirSimAssets/Scripts/Utilities/AirSimSettings.cs)

I believe this dirty fix might help you.

tjxjb, 2019-04-12:
@howard-mahe thanks for the reply! I just tried it and yes I can change the dimensions, great! But I still can't record multiple images at once (like segmentation + scene). Have you managed to do that?

tjxjb, 2019-04-12:
Just to clarify: I can use the Python API to record _consecutive_ frames with different renders, but not the same frame

ironclownfish, 2020-08-14:
@tjxjb Have you tried the simGetImages() API? (The plural version of simGet**Image**)
It can process multiple image requests simultaneously.

closed 2020-09-17 by jonyMarino