microsoft/AirSim #1870: build.cmd - Error with rpclib

San-Holo, 2019-04-04:
Hello,

To describe my situation, I'm trying, for 2 days now, to build AirSim by following th exact steps that are described in the official tutorial. Even if I have already used AirSim once, I didn't encounter the problems I'm having now.

I did these steps in these specific order:

- Register to Epic, download the launcher and the engine
- Install Visual Studio and make sure that VC++ and SDK WIndows 8.1 were selected
- Installed git bash and add it to my path (just a step during the setup, a checkbox actually) to make sure that I could use git command line everywhere, because without this idea, I wasn't able to clone AirSim just in the x64 command prompt for VS 2017

- Start x64 Native Tools Command Prompt for VS 2017
- git clone airsim
- run build.cmd

And then, I had this error:

C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\AirSim>build.cmd
Found cmake version: 3.12.18081601-MSVC_2

********************************************************************************
*********
Downloading rpclib
********************************************************************************
*********
Le terme « iwr » n'est pas reconnu comme nom d'applet de commande, fonction, fichier de script ou programme exécutable. Vérifiez l'orthographe du nom, ou si un chemin d'accès existe, vérifiez que le chemin d'accès est correct et réessayez.
Au niveau de ligne : 1 Caractère : 87
+ & { [Net.ServicePointManager]::SecurityProtocol = [Net.SecurityProtocolType]:
:Tls12; iwr <<<<  https://github.com/rpclib/rpclib/archive/v2.2.1.zip -OutFile ex
ternal\rpclib.zip }
    + CategoryInfo          : ObjectNotFound: (iwr:String) [], CommandNotFound
   Exception
    + FullyQualifiedErrorId : CommandNotFoundException

Le terme « Expand-Archive » n'est pas reconnu comme nom d'applet de commande, f
onction, fichier de script ou programme exécutable. Vérifiez l'orthographe du nom, ou si un chemin d'accès existe, vérifiez que le chemin d'accès est correct
et réessayez.
Au niveau de ligne : 1 Caractère : 19
+ & { Expand-Archive <<<<  -Path external\rpclib.zip -DestinationPath external\
rpclib }
    + CategoryInfo          : ObjectNotFound: (Expand-Archive:String) [], Comm
   andNotFoundException
    + FullyQualifiedErrorId : CommandNotFoundException

Impossible de trouver C:\Program Files (x86)\Microsoft Visual Studio\2017\Commun
ity\AirSim\external\rpclib.zip
Unable to download high-polycount SUV. Your AirSim build will use the default vehicle.

-------------------------------------------------
Here is a screenshot:

[image]

I tried a lot of things, from changing permissions of folders, to execution as administrator. Nothing changed. However, I have identified several words that might help you: **iwr** error, **Expand-archive** error.

I looked for answers from people who seemed to have similar problems, but no answers were useful for my situation. Here is the list of issues I already analyzed : https://github.com/Microsoft/AirSim/issues/1003 , https://github.com/Microsoft/AirSim/issues/1758 ,  https://github.com/Microsoft/AirSim/issues/1178 , https://github.com/Microsoft/AirSim/issues/951

Especially, I tried this one: https://github.com/Microsoft/AirSim/issues/847, it was showing a fix and the concerned lines in the file build.cmd, https://github.com/Microsoft/AirSim/pull/845 . But it changed nothing about my problem. 

Here are all the details of my system:
- Windows 7 Pro
- AirSim v1.2.0
- VS 2017
- Python 3.7.3
- Unreal Engine 4.22.0

Thanks in advance, any help would be perfect !

msb336, 2019-04-08:
@San-Holo 
It looks like your build is failing to download `rpclib.zip`. Not sure why it would fail to download, but try downloading manually from [here](https://github.com/rpclib/rpclib/archive/) and placing into the `AirLib/external` folder.

Relevant `build.cmd` line:
https://github.com/Microsoft/AirSim/blob/affeaf13f453941c9a1c94c496b082eeab2fd238/build.cmd#L59

madratman, 2019-04-08:
@msb336's link has a minor error. You can download the zip file from https://github.com/rpclib/rpclib/archive/v2.2.1.zip
We also observed this in https://github.com/Microsoft/AirSim/issues/1758, not sure why ++

msb336, 2019-04-08:
@madratman updated with proper link

San-Holo, 2019-04-12:
Infortunately, it didn't solve the problem. Nethertheless, I find a solution by looking directly at "iwr" and "Expand-Archive", two things that were not supported by the powershell version I had on my computer.

Upgrading it to the 5.1 version (the last one, if I'm not wrong) solved everything.

Thank you for your help :)

closed 2019-04-12 by San-Holo