microsoft/AirSim #1879: Vehicle goes out of the world in simulation

pranavn248, 2019-04-08:
Hi,

I'm trying to run the hello_drone.py in a new environment that I created. The quadrotor seems to keep going out of the world after the destination is reached. The code I used is almost the same as [hello_drone.py](https://github.com/Microsoft/AirSim/blob/master/PythonClient/multirotor/hello_drone.py) except for the coordinates. It works fine when I give certain coordinates but is acting crazy for a different set of coordinates (which exist within the world). Is this something to do with my environment? I don't have any auto-generated collisions in my world.
I'm using the latest AirSim stable release and the Unreal 14..8.3 version. 
Here's a screenshot of how it looks:
[image]

madratman, 2019-04-09:
hi @pranavn248 , can you share your python script?

pranavn248, 2019-04-17:
@madratman it's pretty much the same as [hello_drone.py](https://github.com/Microsoft/AirSim/blob/master/PythonClient/multirotor/hello_drone.py) except that I've only changed the x,y,z coordinates to a different set of coordinates (something like -15.5527, -5.9786, -8.0225). The drone is able to reach the said coordinates but after it reaches, it goes out of the world like shown in the photo above. Kinda puzzles me as to why that happens.

pranavn248, 2019-07-02:
hey @madratman, have you come up with anything to resolve this issue ? I've been looking to fix it but haven't been able to do so.