microsoft/AirSim #1880: How to adjust Airsim car state update frequency

pengjili, 2019-04-09:
Windows 7, UE4.18, Airsim version: latest, video card: Quadra M620

How can I know or set the car state update frequency in Airsim? Logically the update process of the physics engine should be called every frame(or every tick) according to UE4's default setting. However when I actually print out my car state in approximately 100 fps,  I find out that the value of my car state will be updated within every 60-80ms. So is this the right frequency for updating frame? If I want to achieve higher frequency, how can I do that besides switching to another better computer(e.g. change codes or modify values in UE interface)? Thanks very much for helping!

pengjili, 2019-04-09:
Actually it comes from another problem that no matter how I change the parameter value related to fps (100hz)in sensor param files like ImuSimpleParams.hpp or GpsSimpleParams.hpp etc, the output of sensor values doesn't change. As I look into it I think it's just because of the reason that I mentioned above-car state updating fps isn't high enough, and in fact your sensor values updating fps is mainly up to car states fps.

jonyMarino, 2020-08-14:
@pengjili any update on this issue?

closed 2021-12-31 by jonyMarino