microsoft/AirSim #1884: Any suggestion on setting when packaging into linux on windows ?

beomsik-landingai, 2019-04-11:
After I built a package targeted to Linux from Windows, it sort of runs on linux. But the problem is some of foliage instances in UE, especially ones closer to a vehicle just are not rendered, even with "NoDisplay" mode in subwindows. But instances are there as I could get collision event properly.

 I know this is probably not related to AirSim, but since you guys releases packaged binary to linux, wanted to ask if there's any gotchas you guys found while working on packaging on linux such as lighting settings, other options?

BTW, I had to turn off some static lighting as I got run out of memory issue while packaging. Not sure if it's related. 

Thanks!

madratman, 2019-04-12:
hey, this seems like a material problem. 
Can you check the unreal logs while you packaged the binary, and see if you got any material / shader compiler warnings?

Also, here are a couple of flags worth trying : `-opengl4` and `-vulkan`
For the second option, you might need to install vulkan. in ubuntu, it's just a
`$ sudo apt install vulkan vulkan-utils`
For other distros, see https://linuxconfig.org/install-and-test-vulkan-on-linux

beomsik-landingai, 2019-04-12:
running the application with -opengl4 and -vulkan worked. Thanks, much appreciated!

closed 2019-04-12 by beomsik-landingai

madratman, 2019-04-12:
awesome! we'll add this to the documentation

beomsik-landingai, 2019-04-12:
vulkan tag solved the problem, opengl tag didn't help.