microsoft/AirSim #1885: Problem with tracing, in Computer Vision Mode

JunzheJosephZhu, 2019-04-13:
When using computer vision mode, I used sin(t) for x and cos(t) for y coordinates. But sometimes unreal just breaks and VIsual Studio tells me there's something wrong with tracing. What's the problem?

msb336, 2019-04-15:
@JunzheJosephZhu can you provide a little more information?

When you say you used sin(t) for x and cos(t) for y, what API are you using, and what is your intended functionality?

Also, please attach the full error message to this thread.

jonyMarino, 2020-08-14:
@JunzheJosephZhu  Please feel free to open a new issue if you still have this problem, and include the following (as requested when opening a bug issue)

*Make sure to write all reproduction steps
*Include full error message in text form
*Search issues for error message before filing issue
*Attach screenshot if applicable
*Include code to run if applicable

Thanks!

closed 2020-08-14 by jonyMarino