microsoft/AirSim #1886: Imitation learning/train_model.py

Spatialghost, 2019-04-14:
Hi, i train the model, but after few epochs i get error:
Traceback (most recent call last):
  File "train_model.py", line 94, in <module>
    validation_data=eval_generator, validation_steps=num_eval_examples//batch_size, verbose=2)
  File "C:\Python36\lib\site-packages\keras\legacy\interfaces.py", line 87, in wrapper
    return func(*args, **kwargs)
  File "C:\Python36\lib\site-packages\keras\engine\training.py", line 2169, in fit_generator
    use_multiprocessing=use_multiprocessing)
  File "C:\Python36\lib\site-packages\keras\legacy\interfaces.py", line 87, in wrapper
    return func(*args, **kwargs)
  File "C:\Python36\lib\site-packages\keras\engine\training.py", line 2305, in evaluate_generator
    raise ValueError('Received an empty batch. '
ValueError: Received an empty batch. Batches should at least contain one item.

Please help, how to solve this problem?

NextSim, 2019-04-15:
Your best bet is to ask this in the autonomous driving cookbook repository

[https://github.com/Microsoft/AutonomousDrivingCookbook/issues](url)

jonyMarino, 2020-08-14:
closed due to age and does not seem to be an airsim issue

closed 2020-08-14 by jonyMarino