microsoft/AirSim #1891: could not find function 'getCarState'

habibutsu, 2019-04-16:
After downloading version `v1.2.0 - Linux/ Blocks.zip` and running following code:

```
client = CarClient()
client.confirmConnection()
client.enableApiControl(True)
client.getCarState()
```

I getting following error

`RPCError: rpclib: server could not find function 'getCarState' with argument count 1.`

msb336, 2019-04-16:
@habibutsu This is an interesting error. At first try, I am not able to replicate it. Which Linux operating system are you using?

habibutsu, 2019-04-17:
@msb336  Ubuntu 16.04.4 LTS (GNU/Linux 4.15.0-46-generic x86_64)

habibutsu, 2019-04-25:
Maybe I can enable debug mode, verbose level for logging or something else for investigate this problem?

jonyMarino, 2020-08-17:
@habibutsu do you still have this issue with the new airsim version?

closed 2020-09-02 by jonyMarino