microsoft/AirSim #1893: Position of the car with respect to the environment

HsnAtwi, 2019-04-17:
Hello, I'm trying to train a car in the NH environment (Linux version with Unreal Engine) using Reinforcement Learning.

I need to know the position of the car wrt the road for example, I've used the method in the compute_reward function here https://github.com/Microsoft/AirSim/blob/fff500f6111d463c6f8a0cf386d54cfccf5c0d1b/PythonClient/car/DQNcar.py#L468
but it's not clear on what basis it calculates the distance between the edge of the road and the car itself, the threshold distance (=3.5) is not clear as well and it's not consistent on all the roads. There is also no explanation on what are the points used or what do they represent 

I've also tried the method in the driving cookbook-- distributed RL but also did not work.

So, is there a way to know the map of the NH containing the coordinates of the roads for example, or at least an explanation to the 9 points used to calculate the distance and the chose behind the threshold distance.
This will also benefit me when I need to try different initial positions for the car to start training from

Any help would be appreciated.

BrianN92, 2019-04-17:
Thanks for posting! I am learning the distributed RL tutorial and I have the same question. It seems like the figure of precomputed centers of each road is not as same as the NH map. The vehicle does not go with those lines in the same direction.

NextSim, 2019-04-26:
You can use the files reward_points.txt	and road_lines.txt to look at the road paths and the reward points.. you can plot the points and see it in excel.

Remember that the autonomous driving cookbook uses a specific version of the neighborhood map using an older, incompatible version of the airsim api.  It has different coordinates, so the road lines on the autonomous driving cookbook map will not be the same if you mapped with the latest released neighborhood map.

It is also probably best to post in the autonomous driving cookbook issues.

HsnAtwi, 2019-04-26:
Thanks for the reply @NextSim 
Yeah I expected that they won't be the same, I wanna know if there is a defined map or coordinates for the latest version on Linux

MoritzGa, 2019-05-29:
I'm working on Reinforcement Learning as well and tried using the given AirSim code. I believe the given  given points that are defined [here](https://github.com/Microsoft/AirSim/blob/fff500f6111d463c6f8a0cf386d54cfccf5c0d1b/PythonClient/car/DQNcar.py#L468) are simply the center of the road. Hence it will only work in the NH environment.

zimmy87, 2021-01-07:
possibly related to #1009

zimmy87, 2021-01-15:
Hi, thank you for reporting this issue.  This script was recently removed in #3215 and replaced with a new script at \AirSim\PythonClient\reinforcement_learning\dqn_car.py that uses the new airgym module. I tested the new script locally and it runs for me. Can you retest with the new script and report back if you see any issues?

closed 2021-01-15 by zimmy87

LLM1602, 2021-07-05:
> You can use the files reward_points.txt and road_lines.txt to look at the road paths and the reward points.. you can plot the points and see it in excel.
>
> Remember that the autonomous driving [...]

i want to konw where can find the file which you said  reward_points.txt and road_lines.txt , can you told me the position of the file in detail?Thank you very much