microsoft/AirSim #1894: Building Blocks Environment

vishal-prgmr, 2019-04-17:
Platform: Windows 7
Visual Studio: 2017
Unreal Editor: 4.18

Problem: build.cmd gives a sucessful build but doesnt generate the .sln under the Unreal\Environment\Blocks directory.

When generate_from_git.bat is run, the following error pops up. (Also when the Blocks.uproject file is right click and "Generate Project Files" is chosen)

Error message in the pop-up
" Running C:/Program Files/Epic Games/UE_4.18/Engine/Binaries/DotNET/UnrealBuildTool.exe  -projectfiles -project="C:/AirSim-master/Unreal/Environments/Blocks/Blocks.uproject" -game -rocket -progress -log="C:\AirSim-master\Unreal\Environments\Blocks/Saved/Logs/UnrealVersionSelector-2019.04.17-15.41.31.log"
Discovering modules, targets and source code for project...
While compiling C:\AirSim-master\Unreal\Environments\Blocks\Intermediate\Build\BuildRules\BlocksModuleRules.dll:
ERROR: c:\AirSim-master\Unreal\Environments\Blocks\Plugins\AirSim\Source\AirSim.Build.cs(50,17) : error CS0103: The name 'PublicDefinitions' does not exist in the current context
ERROR: c:\AirSim-master\Unreal\Environments\Blocks\Plugins\AirSim\Source\AirSim.Build.cs(51,17) : error CS0103: The name 'PublicDefinitions' does not exist in the current context
ERROR: c:\AirSim-master\Unreal\Environments\Blocks\Plugins\AirSim\Source\AirSim.Build.cs(55,17) : error CS0103: The name 'PublicDefinitions' does not exist in the current context
ERROR: c:\AirSim-master\Unreal\Environments\Blocks\Plugins\AirSim\Source\AirSim.Build.cs(61,17) : error CS0103: The name 'PublicDefinitions' does not exist in the current context
ERROR: c:\AirSim-master\Unreal\Environments\Blocks\Plugins\AirSim\Source\AirSim.Build.cs(83,9) : error CS0103: The name 'PublicDefinitions' does not exist in the current context
ERROR: c:\AirSim-master\Unreal\Environments\Blocks\Plugins\AirSim\Source\AirSim.Build.cs(84,9) : error CS0103: The name 'PublicDefinitions' does not exist in the current context
ERROR: c:\AirSim-master\Unreal\Environments\Blocks\Plugins\AirSim\Source\AirSim.Build.cs(85,9) : error CS0103: The name 'PublicDefinitions' does not exist in the current context
ERROR: c:\AirSim-master\Unreal\Environments\Blocks\Plugins\AirSim\Source\AirSim.Build.cs(154,9) : error CS0103: The name 'PublicDefinitions' does not exist in the current context
ERROR: UnrealBuildTool Exception: Unable to compile source files. "

[image]

msb336, 2019-04-17:
There are a couple things you can do to troubleshoot:
1. **Generate project files from inside unreal**
  - open `Blocks.uproject` in the unreal editor
  - In the editor `File -> Refresh Visual Studio Project`
2. **Reset AirSim plugin**
  - Delete `Intermediate`, `Binaries`, and `Plugins\AirSim` folders in the project directory
  - Copy `Unreal\Plugins\AirSim` to the blocks project plugins folder
  - Generate the VS solution using the steps in part 1
3. **Clean out the repository and rebuild**
  - This will remove any files that are not versioned in your repo, so make sure to commit your work before doing this!
  - run `git clean -fdx` from the command line
  - Rebuild airsim, and add the AirSim Plugin folder back to the Blocks environment

vishal-prgmr, 2019-04-18:
@msb336 I tried the steps you listed. Unfortunately nothing helped. 
Thank You

msb336, 2019-04-18:
@vishal-prgmr It appears this is from the most recent PR to be merged, which feature some definition changes to allow airsim to work with versions 4.19+.

For an immediate solution, you can roll back to the previous commit to the repo:
`git checkout c494e2c15ec6f363a60a7ca6b6640f3394f528e2`

@sytelus there is a [new pr which fixes this issue](https://github.com/Microsoft/AirSim/pull/1897), please merge it when you can.

CurryTang, 2019-04-19:
@vishal-prgmr I've encountered the same problem. It seems that it's related to the updates of some API of Unreal Build Tools. You can check https://forums.unrealengine.com/development-discussion/c-gameplay-programming/1527721-error-cs0103-the-name-publicdefinitions-does-not-exist-in-the-current-context

msb336, 2019-04-19:
@CurryTang you are correct. [this pr](https://github.com/Microsoft/AirSim/pull/1897) should remedy the issue. Please update your master branch and try building the blocks project again.

vishal-prgmr, 2019-04-23:
@msb336 , I will get back after trying this out. Thank you so much.

vishal-prgmr, 2019-04-23:
@msb336 , Got it working. Thank You.

closed 2019-04-23 by vishal-prgmr