microsoft/AirSim #1898: Compile on MacOS

Tongbh, 2019-04-18:
I'm new to Airsim, and my OS is macOS Mojave 10.14.4, when I enter ./setup.sh in the terminal, at the step of cmake --build, some bugs may had happened. What can I do? 
tongbohandeMacBook-Pro:AirSim-master tongbohan$ ./setup.sh
+ set -e
+++ dirname ./setup.sh
++ cd .
++ pwd
+ SCRIPT_DIR=/Users/tongbohan/Desktop/AirSim-master
+ pushd /Users/tongbohan/Desktop/AirSim-master
+ downloadHighPolySuv=true
+ [[ '' == \-\-\n\o\-\f\u\l\l\-\p\o\l\y\-\c\a\r ]]
++ uname
+ '[' Darwin == Darwin ']'
+ [[ ! -z '' ]]
+ brew tap llvm-hs/homebrew-llvm
Updating Homebrew...
^C+ brew install llvm-5.0
Updating Homebrew...
^C==> Installing llvm-5.0 from llvm-hs/llvm
==> Downloading http://releases.llvm.org/5.0.2/llvm-5.0.2.src.tar.xz
Already downloaded: /Users/tongbohan/Library/Caches/Homebrew/downloads/1f991e7a96992598f4d76e4ace578bef379f838683f63f18819471b76e7ce89e--llvm-5.0.2.src.tar.xz
==> Downloading http://releases.llvm.org/5.0.2/cfe-5.0.2.src.tar.xz
Already downloaded: /Users/tongbohan/Library/Caches/Homebrew/downloads/d315262ba3613abde4cdda1d330565263a7dfab278f059ae1049b48a5fdb146d--cfe-5.0.2.src.tar.xz
==> Downloading http://releases.llvm.org/5.0.2/libcxx-5.0.2.src.tar.xz
Already downloaded: /Users/tongbohan/Library/Caches/Homebrew/downloads/733192995fcd980375579f7d38558c568b372692e782b90dbe849426017e862c--libcxx-5.0.2.src.tar.xz
==> Downloading http://releases.llvm.org/5.0.2/lld-5.0.2.src.tar.xz
Already downloaded: /Users/tongbohan/Library/Caches/Homebrew/downloads/5c38a1148c1f7c7f3574bd8fdb31a1d203965b6668f0a6d8465c7b19912f53e9--lld-5.0.2.src.tar.xz
==> Downloading http://releases.llvm.org/5.0.2/polly-5.0.2.src.tar.xz
Already downloaded: /Users/tongbohan/Library/Caches/Homebrew/downloads/342527ed42b46bac64443c05a2a0b1123e78d6fedd3564e37edb5cdf7dd50ca1--polly-5.0.2.src.tar.xz
==> Downloading http://releases.llvm.org/5.0.2/clang-tools-extra-5.0.2.src.tar.x
Already downloaded: /Users/tongbohan/Library/Caches/Homebrew/downloads/df4108bd0abc5eb490ffc382ad560ce52def0baeb0a25e522e657ebf1547de26--clang-tools-extra-5.0.2.src.tar.xz
==> Downloading http://releases.llvm.org/5.0.2/openmp-5.0.2.src.tar.xz
Already downloaded: /Users/tongbohan/Library/Caches/Homebrew/downloads/6dd30d3b3ab03df8a491be48e7920f6974bb03dc7b868290ef4d01734d30802e--openmp-5.0.2.src.tar.xz
==> Downloading http://releases.llvm.org/5.0.2/libcxx-5.0.2.src.tar.xz
Already downloaded: /Users/tongbohan/Library/Caches/Homebrew/downloads/733192995fcd980375579f7d38558c568b372692e782b90dbe849426017e862c--libcxx-5.0.2.src.tar.xz
==> Downloading http://releases.llvm.org/5.0.2/libunwind-5.0.2.src.tar.xz
Already downloaded: /Users/tongbohan/Library/Caches/Homebrew/downloads/a46542902de7fbb51111d11d39c1bc3f74b950173cf5b68fcdd9956aaed12381--libunwind-5.0.2.src.tar.xz
==> Downloading http://releases.llvm.org/5.0.2/compiler-rt-5.0.2.src.tar.xz
Already downloaded: /Users/tongbohan/Library/Caches/Homebrew/downloads/afca2ed77a2caf3419832039807dbb6b170f845421d512d5a8588853cd55a53c--compiler-rt-5.0.2.src.tar.xz
==> cmake /private/tmp/llvm-5.0-20190418-16134-17p8358/llvm-5.0.2.src -DCMAKE_C_
==> cmake --build .
Last 15 lines from /Users/tongbohan/Library/Logs/Homebrew/llvm-5.0/02.cmake:
cd /tmp/llvm-5.0-20190418-16134-1b2oql3/lib/CodeGen && /usr/local/Homebrew/Library/Homebrew/shims/mac/super/clang++  -DLLVM_BUILD_GLOBAL_ISEL -D_DEBUG -D__STDC_CONSTANT_MACROS -D__STDC_FORMAT_MACROS -D__STDC_LIMIT_MACROS -I/tmp/llvm-5.0-20190418-16134-1b2oql3/lib/CodeGen -I/tmp/llvm-5.0-20190418-16134-17p8358/llvm-5.0.2.src/lib/CodeGen -I/tmp/llvm-5.0-20190418-16134-1b2oql3/include -I/tmp/llvm-5.0-20190418-16134-17p8358/llvm-5.0.2.src/include  -stdlib=libc++ -fPIC -fvisibility-inlines-hidden -Werror=date-time -std=c++11 -Wall -W -Wno-unused-parameter -Wwrite-strings -Wcast-qual -Wmissing-field-initializers -pedantic -Wno-long-long -Wcovered-switch-default -Wnon-virtual-dtor -Wdelete-non-virtual-dtor -Wstring-conversion   -isysroot /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk   -UNDEBUG   -o CMakeFiles/LLVMCodeGen.dir/TwoAddressInstructionPass.cpp.o -c /tmp/llvm-5.0-20190418-16134-17p8358/llvm-5.0.2.src/lib/CodeGen/TwoAddressInstructionPass.cpp
[ 50%] Building CXX object lib/CodeGen/CMakeFiles/LLVMCodeGen.dir/UnreachableBlockElim.cpp.o
cd /tmp/llvm-5.0-20190418-16134-1b2oql3/lib/CodeGen && /usr/local/Homebrew/Library/Homebrew/shims/mac/super/clang++  -DLLVM_BUILD_GLOBAL_ISEL -D_DEBUG -D__STDC_CONSTANT_MACROS -D__STDC_FORMAT_MACROS -D__STDC_LIMIT_MACROS -I/tmp/llvm-5.0-20190418-16134-1b2oql3/lib/CodeGen -I/tmp/llvm-5.0-20190418-16134-17p8358/llvm-5.0.2.src/lib/CodeGen -I/tmp/llvm-5.0-20190418-16134-1b2oql3/include -I/tmp/llvm-5.0-20190418-16134-17p8358/llvm-5.0.2.src/include  -stdlib=libc++ -fPIC -fvisibility-inlines-hidden -Werror=date-time -std=c++11 -Wall -W -Wno-unused-parameter -Wwrite-strings -Wcast-qual -Wmissing-field-initializers -pedantic -Wno-long-long -Wcovered-switch-default -Wnon-virtual-dtor -Wdelete-non-virtual-dtor -Wstring-conversion   -isysroot /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk   -UNDEBUG   -o CMakeFiles/LLVMCodeGen.dir/UnreachableBlockElim.cpp.o -c /tmp/llvm-5.0-20190418-16134-17p8358/llvm-5.0.2.src/lib/CodeGen/UnreachableBlockElim.cpp
[ 50%] Building CXX object lib/CodeGen/CMakeFiles/LLVMCodeGen.dir/VirtRegMap.cpp.o
cd /tmp/llvm-5.0-20190418-16134-1b2oql3/lib/CodeGen && /usr/local/Homebrew/Library/Homebrew/shims/mac/super/clang++  -DLLVM_BUILD_GLOBAL_ISEL -D_DEBUG -D__STDC_CONSTANT_MACROS -D__STDC_FORMAT_MACROS -D__STDC_LIMIT_MACROS -I/tmp/llvm-5.0-20190418-16134-1b2oql3/lib/CodeGen -I/tmp/llvm-5.0-20190418-16134-17p8358/llvm-5.0.2.src/lib/CodeGen -I/tmp/llvm-5.0-20190418-16134-1b2oql3/include -I/tmp/llvm-5.0-20190418-16134-17p8358/llvm-5.0.2.src/include  -stdlib=libc++ -fPIC -fvisibility-inlines-hidden -Werror=date-time -std=c++11 -Wall -W -Wno-unused-parameter -Wwrite-strings -Wcast-qual -Wmissing-field-initializers -pedantic -Wno-long-long -Wcovered-switch-default -Wnon-virtual-dtor -Wdelete-non-virtual-dtor -Wstring-conversion   -isysroot /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk   -UNDEBUG   -o CMakeFiles/LLVMCodeGen.dir/VirtRegMap.cpp.o -c /tmp/llvm-5.0-20190418-16134-17p8358/llvm-5.0.2.src/lib/CodeGen/VirtRegMap.cpp
[ 50%] Building CXX object lib/CodeGen/CMakeFiles/LLVMCodeGen.dir/WinEHPrepare.cpp.o
cd /tmp/llvm-5.0-20190418-16134-1b2oql3/lib/CodeGen && /usr/local/Homebrew/Library/Homebrew/shims/mac/super/clang++  -DLLVM_BUILD_GLOBAL_ISEL -D_DEBUG -D__STDC_CONSTANT_MACROS -D__STDC_FORMAT_MACROS -D__STDC_LIMIT_MACROS -I/tmp/llvm-5.0-20190418-16134-1b2oql3/lib/CodeGen -I/tmp/llvm-5.0-20190418-16134-17p8358/llvm-5.0.2.src/lib/CodeGen -I/tmp/llvm-5.0-20190418-16134-1b2oql3/include -I/tmp/llvm-5.0-20190418-16134-17p8358/llvm-5.0.2.src/include  -stdlib=libc++ -fPIC -fvisibility-inlines-hidden -Werror=date-time -std=c++11 -Wall -W -Wno-unused-parameter -Wwrite-strings -Wcast-qual -Wmissing-field-initializers -pedantic -Wno-long-long -Wcovered-switch-default -Wnon-virtual-dtor -Wdelete-non-virtual-dtor -Wstring-conversion   -isysroot /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk   -UNDEBUG   -o CMakeFiles/LLVMCodeGen.dir/WinEHPrepare.cpp.o -c /tmp/llvm-5.0-20190418-16134-17p8358/llvm-5.0.2.src/lib/CodeGen/WinEHPrepare.cpp
[ 50%] Building CXX object lib/CodeGen/CMakeFiles/LLVMCodeGen.dir/XRayInstrumentation.cpp.o
cd /tmp/llvm-5.0-20190418-16134-1b2oql3/lib/CodeGen && /usr/local/Homebrew/Library/Homebrew/shims/mac/super/clang++  -DLLVM_BUILD_GLOBAL_ISEL -D_DEBUG -D__STDC_CONSTANT_MACROS -D__STDC_FORMAT_MACROS -D__STDC_LIMIT_MACROS -I/tmp/llvm-5.0-20190418-16134-1b2oql3/lib/CodeGen -I/tmp/llvm-5.0-20190418-16134-17p8358/llvm-5.0.2.src/lib/CodeGen -I/tmp/llvm-5.0-20190418-16134-1b2oql3/include -I/tmp/llvm-5.0-20190418-16134-17p8358/llvm-5.0.2.src/include  -stdlib=libc++ -fPIC -fvisibility-inlines-hidden -Werror=date-time -std=c++11 -Wall -W -Wno-unused-parameter -Wwrite-strings -Wcast-qual -Wmissing-field-initializers -pedantic -Wno-long-long -Wcovered-switch-default -Wnon-virtual-dtor -Wdelete-non-virtual-dtor -Wstring-conversion   -isysroot /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk   -UNDEBUG   -o CMakeFiles/LLVMCodeGen.dir/XRayInstrumentation.cpp.o -c /tmp/llvm-5.0-20190418-16134-17p8358/llvm-5.0.2.src/lib/CodeGen/XRayInstrumentatio [... cut ...] cpp.o CMakeFiles/LLVMCodeGen.dir/TargetSubtargetInfo.cpp.o CMakeFiles/LLVMCodeGen.dir/TwoAddressInstructionPass.cpp.o CMakeFiles/LLVMCodeGen.dir/UnreachableBlockElim.cpp.o CMakeFiles/LLVMCodeGen.dir/VirtRegMap.cpp.o CMakeFiles/LLVMCodeGen.dir/WinEHPrepare.cpp.o CMakeFiles/LLVMCodeGen.dir/XRayInstrumentation.cpp.o 
[ 50%] Built target LLVMCodeGen
make: *** [all] Error 2

If reporting this issue please do so at (not Homebrew/brew or Homebrew/core):
  https://github.com/llvm-hs/homebrew-llvm/issues

klintan, 2019-04-24:
LLVM-5.0 is really old, using Brew I installed the latest one `brew install llvm` 

I had a bunch of other issues, which I don't know if they are solved yet but just comment out line 32 in the `setup.sh` file and run. 

I also had to run `open /Library/Developer/CommandLineTools/Packages/` and install the package `macOS_SDK_headers_for_macOS_10.14.pkg` otherwise the cmake bootstrap step fails as described here:
https://stackoverflow.com/questions/52509602/cant-compile-c-program-on-a-mac-after-upgrade-to-mojave

Not sure I'll succeed yet, but at least the `setup.sh` is running now.

Tongbh, 2019-04-25:
Thanks. I succeed in Windows, the Airsim developed by Microsoft seemed a little unfriendly to OSX.

klintan, 2019-04-26:
I got it to run now on OSX (latest Mojave) see https://github.com/Microsoft/AirSim/pull/1914

as described in my comment above I also changed to the latest llvm in most places (build.sh and setup.sh + Unity/build.sh). It also works using `gcc` and `g++` as compilers. 

I haven't put them in a PR yet, because I think that is a bigger change than the ones in the referenced PR above (because it is pretty much only affecting MacOSX building/compiling)

vephinx, 2019-10-24:
I am running MacOS Mojave, and I still get the same error that the pull request was supposed to fix:

```
$ ./setup.sh
+++ dirname ./setup.sh
++ cd .
++ pwd
+ SCRIPT_DIR=/Users/sebastian/Projects/airsim/AirSim
+ pushd /Users/sebastian/Projects/airsim/AirSim
+ downloadHighPolySuv=true
+ gccBuild=false
+ MIN_CMAKE_VERSION=3.10.0
+ MIN_GCC_VERSION=6.0.0
+ [[ 0 -gt 0 ]]
+ false
++ uname
+ '[' Darwin == Darwin ']'
+ brew update
Already up-to-date.
+ brew tap llvm-hs/homebrew-llvm
+ brew install llvm-5.0

...

[ 53%] Built target LLVMCodeGen
make: *** [all] Error 2

If reporting this issue please do so at (not Homebrew/brew or Homebrew/core):
  https://github.com/llvm-hs/homebrew-llvm/issues

+ export C_COMPILER=/usr/local/opt/llvm-5.0/bin/clang-5.0
+ C_COMPILER=/usr/local/opt/llvm-5.0/bin/clang-5.0
+ export COMPILER=/usr/local/opt/llvm-5.0/bin/clang++-5.0
+ COMPILER=/usr/local/opt/llvm-5.0/bin/clang++-5.0
```

kraftpunk97, 2019-11-09:
bumping

klintan, 2020-01-18:
> bumping
> @klintan quickly glancing at the changes in the PR you linked, it seems that `setup.sh` still uses llvm-5.0, whereas the latest version as of writing this comment is 9.0? It's unclear whether you used v5.0 or a newer release.

Thanks for your patience. Yeah I didn't push that because I didn't know if it was just me, and I think I have a comment about that in some issue/PR.
However I did use a newer version of llvm.

jonyMarino, 2020-09-04:
Closed due to age.
Please feel free to open a new issue (referring this one) if you still have this problem, and include the following (as requested when opening a bug issue)

*Make sure to write all reproduction steps
*Include full error message in text form
*Search issues for error message before filing issue
*Attach screenshot if applicable
*Include code to run if applicable

Thanks!

closed 2020-09-04 by jonyMarino