microsoft/AirSim #1907: Can i get actual distance to the obstacle by the depth image？

machenxiang, 2019-04-24:
I have transformed the pfm image to the depth image. but I want to get the actual distance to the obstacle. I want to ask how to do it?

tejaswid, 2019-04-28:
If you are using the DepthPlanner image, the depth value at each pixel is the depth of the object in the camera frame. For every pixel (u,v) in the DepthPlanner image, the true point (x, y, z) in the world in the camera frame of reference can be calculated as
x = (u - cx) * d / fx
y = (v - cy) * d / fy
z = d

where 
d is the value at the pixel (u,v) in the DepthPlanner image,
(cx, cy) are the principal points = (width/2, height/2)
(fx, fy) are the focal lengths.

For the values (cx, cy, fx, fy) refer to your settings.json file. The focal lengths can be calculated using the FOV value.

machenxiang, 2019-05-07:
@tejaswid  thanks for the reply. i will try this

madratman, 2019-05-14:
Another option would be to use [depth_image_proc](https://wiki.ros.org/depth_image_proc) over ROS. 
I've added instructions here [here](https://github.com/microsoft/AirSim/tree/master/ros/src/airsim_tutorial_pkgs#single-drone-with-monocular-and-depth-cameras-and-lidar). 

But yeah, if you don't want the ros dependency, it's basically camera unprojection  / what @tejaswid wrote

closed 2019-05-14 by madratman