microsoft/AirSim #1908: getPointCloud - method's implementation

vishal-prgmr, 2019-04-24:
Greetings,

Could someone please tell where to look for the implementation of the **getPointCloud** method which is in **LidarSimple**.**hpp** ? 

Thank You :)

msb336, 2019-04-24:
@visahl-prgmr `getPointCloud` is an internal call to unreal, the client call to get lidar data is `getLidarData()`.

If you are just looking to retrieve lidar data, use `getLidarData()`

If you are curious as to how AirSim gets it's pointcloud data from unreal, `getPointCloud` is defined here:
https://github.com/Microsoft/AirSim/blob/1fd6a3fc311c704bbbb0b2b6b245e8fa0ba26c8b/Unreal/Plugins/AirSim/Source/UnrealSensors/UnrealLidarSensor.cpp#L44

vishal-prgmr, 2019-04-25:
@msb336 , Thank you for the reply. I am looking forward to see how the pointcloud is generated inorder to semantically label them. Do you have any heads up for me in this? 

Thank you once again.

closed 2019-04-26 by vishal-prgmr