microsoft/AirSim #1912: getLidarData() doesn't work well in Blocks and cunstom environment

orangeccc, 2019-04-25:
I use the same simple test code for collecting lidar data and vehicle trajectory as below:

        i=0
        traj = []
        while(i<250):
            lidarData = client.getLidarData()
            car_state = client.getCarState()
            points = parse_lidarData(lidarData)
            numpy.savetxt("points\{}.csv".format(i),points)
            a = car_state.kinematics_estimated.position
            traj.append([a.x_val, a.y_val, a.z_val, car_state.timestamp]r)
            i += 1
            time.sleep(0.1)
        numpy.savetxt("trajectory.csv", traj)

  But I find it works differently in different environment. When using CityEnviron , the center of the point cloud changes correctly as the vehicle moves , so I can get a point cloud scene of the environment by merging all points clouds as shown in Fig 1. The point cloud centers follows the trajectory (red points ) well.
[image]
 Fig 1 lidar points in CityEnviron

  However , the center of the point cloud doesn't change as the vehicle moves when using Blocks or other custom environment ，Fig 2. It seems something wrong with getLidarData().
[image]
Fig 2 lidar points in Blocks

  Maybe it is similar with #1865 , but I'm not quite sure.

  Thanks for any help!

* Windows 10, AirSim 1.2.0, Python 3.6.3, Unreal 4.19

vishal-prgmr, 2019-04-25:
@orangeccc , What is platform used for visualization?

orangeccc, 2019-04-25:
@vishal-prgmr  CloudCompare. See https://github.com/CloudCompare/CloudCompare

msb336, 2019-04-25:
@vishal-prgmr 
Can you copy and paste your settings.json file used? There is an option to specify which frame you want LiDAR to return in (global, or sensor local). You can see more [here](https://github.com/Microsoft/AirSim/blob/master/docs/lidar.md#lidar-configuration).

orangeccc, 2019-04-26:
@msb336 Thanks! I used the SensorLocalFrame for DataFrame Option  , now I change to VehicleInertialFrame and it works. 
[image]

closed 2019-04-26 by orangeccc

lcyawym, 2019-04-30:
@orangeccc hello， I have some errors on get lidar data，can you show your setting flie？When I simulation it always happen that no vehicle named  or no lidar on vehicle 。 I think maybe something wrong on setting files。 thankyou

vishal-prgmr, 2019-04-30:
@lcyawym 

May be this is helpful

{
    "SeeDocsAt": "https://github.com/Microsoft/AirSim/blob/master/docs/settings_json.md",
    "SettingsVersion": 1.2,

    "SimMode": "Car",
	"RecordUIVisible": true,
	"LogMessagesVisible": true,
	"ViewMode": "",
	"Recording": {
    "RecordOnMove": false,
    "RecordInterval": 0.05,
    "Cameras": [
        { "CameraName": "4", "ImageType": 0, "PixelsAsFloat": false, "Compress": true },
		{ "CameraName": "4", "ImageType": 3, "PixelsAsFloat": false, "Compress": true }
    ]
	
	},
	"CameraDefaults": { 
	"CaptureSettings": [
      {
        "ImageType": 0,
        "Width": 1280,
        "Height": 960,
        "FOV_Degrees": 90
       },
	   {
        "ImageType": 3,
        "Width": 1280,
        "Height": 960,
        "FOV_Degrees": 90
       },
	   {
        "ImageType": 5,
        "Width": 1280,
        "Height": 960,
        "FOV_Degrees": 90
       }
	   
    ]
	},
    
	"SubWindows": [
					{"WindowID": 0, "ImageType": 0, "CameraName": "front_custom", "Visible": true},
					{"WindowID": 1, "ImageType": 3, "CameraName": "front_custom", "Visible": true},
					{"WindowID": 2, "ImageType": 5, "CameraName": "front_custom", "Visible": true}
				  ],
     "Vehicles": {
        "EgoCar": {
            "VehicleType": "PhysXCar",
            "AutoCreate": true,
			"Cameras": {
				"front_custom": {
				"CaptureSettings": [
				  {
					"ImageType": 0,
					"Widthidth": 1280,
					"Height": 960,
					"FOV_Degrees": 90
					
				  },
				  {
					"ImageType": 3,
					"Widthidth": 1280,
					"Height": 960,
					"FOV_Degrees": 90
					
				  },
				  {
					"ImageType": 5,
					"Width": 1280,
					"Height": 960,
					"FOV_Degrees": 90
					
				  }
				],
					
				
				"X": 0, "Y": 0, "Z": -1.9,
				"Pitch": 0, "Roll": 0, "Yaw": 45    
				}
			  },
			
			"Sensors": {
                "LidarSensor1": { 
                    "SensorType": 6,
                    "Enabled" : true,
                    "NumberOfChannels": 16,
					"Range": 100,
                    "RotationsPerSecond": 10,
                    "PointsPerSecond": 100000,
                    "X": 0, "Y": 0, "Z": -1.9,
                    "Roll": 0, "Pitch": 0, "Yaw" : 0,
                    "VerticalFOVUpper": 10,
                    "VerticalFOVLower": -10,
                    "HorizontalFOVStart": -360,
                    "HorizontalFOVEnd": 360,
                    "DrawDebugPoints": true,
                    "DataFrame": "VehicleInertialFrame"
                }		
            }
        }
    }
}