microsoft/AirSim #1913: segmentation bug

tchaton, 2019-04-25:
[image]
``
import airsim
import json
import time
import os
import numpy as np
from datetime import datetime
import copy

client.simSetSegmentationObjectID("[\w]*", -1, True)

for i in range(3, 10):
    print(i)
    print(client.simSetSegmentationObjectID('Car{}'.format(i), 10 * i))
``

``
{
	"SettingsVersion": 1.2,
	"SimMode": "Car",

	"OriginGeopoint": {
	"Latitude": 5220.82,
	"Longitude": -3556.53,
	"Altitude": 138.68
	},

	"CameraDefaults": {
		"CaptureSettings": [{
			"ImageType": 0,
			"Width": 1024,
			"Height": 576
		},
		{
			"ImageType": 3,
			"Width": 1024,
			"Height": 576
		},
		{
			"ImageType": 5,
			"Width": 1024,
			"Height": 576
		},
		{
			"ImageType": 1,
			"Width": 1024,
			"Height": 576
		}]
	},

	"SubWindows": [{
			"WindowID": 0,
			"CameraName": "0",
			"ImageType": 0,
			"Visible": true
		},
		{
			"WindowID": 1,
			"CameraName": "0",
			"ImageType": 3,
			"Visible": true
		},
		{
			"WindowID": 2,
			"CameraName": "0",
			"ImageType": 5,
			"Visible": true
		}
	],
	"SegmentationSettings": {
		"InitMethod": "CommonObjectsRandomIDs",
		"MeshNamingMethod": "OwnerName",
		"OverrideExisting": false
},

	"Vehicles": {
		"Car1": {
			"VehicleType": "PhysXCar",
			"X": -25,
			"Y": 0,
			"Z": -4,
			"Pitch": 0,
			"Roll": 0,
			"Yaw": 0
		},
		"Car2": {
			"VehicleType": "PhysXCar",
			"X": 0,
			"Y": 25,
			"Z": -4,
			"Pitch": 0,
			"Roll": 0,
			"Yaw": 270
		},
		"Car3": {
			"VehicleType": "PhysXCar",
			"X": 15,
			"Y": 0,
			"Z": -4,
			"Pitch": 0,
			"Roll": 0,
			"Yaw": 180
		},
		"Car4": {
			"VehicleType": "PhysXCar",
			"X": 0,
			"Y": -15,
			"Z": -4,
			"Pitch": 0,
			"Roll": 0,
			"Yaw": 90
		},
		"Car5": {
			"VehicleType": "PhysXCar",
			"X": 0,
			"Y": -20,
			"Z": -4,
			"Pitch": 0,
			"Roll": 0,
			"Yaw": 45
		},
		"Car6": {
			"VehicleType": "PhysXCar",
			"X": 0,
			"Y": 0,
			"Z": -4,
			"Pitch": 0,
			"Roll": 0,
			"Yaw": 45
		}
	}
}

``

tchaton, 2019-04-25:
In SubView from Unreal, everything is fine. However, when I get the segmentation, I lose one car.

AloshkaD, 2019-09-22:
@tchaton have you figured out what's wrong with the segmentation. I have one additional drone that created and it's not seen in the segmentation view.