microsoft/AirSim #1915: Question: How to ensure the frequency of getting images?

Hezzzzzy, 2019-04-26:
I have tried the python api simGetImages()
but I found this api cost 0.08 seconds,
so I don't know the recording mode(by using the key R) how to ensure the frequency of getting images?

stavBodik, 2019-04-29:
write your own thread which will run and take the results according to the FPS you need CPP : 

`void recordImagesThread()
{
	TTimePoint prevImageTime = 0;
	int i = 0;
	t1 = std::chrono::steady_clock::now();

	while (!isStopRecording) 
	{
		steady_clock::time_point t2 = steady_clock::now();
		auto timePassed = duration_cast<milliseconds>(t2 - t1).count();

		//wait for FPS time to pass
		while (timePassed < 1000.0/FPS)
		{
			t2 = steady_clock::now();
			timePassed = duration_cast<milliseconds>(t2 - t1).count();
			//cout << "waiting" << endl;
		}
		t1 = t2;

		//std::cout << std::to_string(client.getMultirotorState().gps_location.longitude) + "," + std::to_string(client.getMultirotorState().gps_location.latitude) + "," + std::to_string(client.getMultirotorState().gps_location.altitude) << endl;
		std::vector<ImageRequest> request = { ImageRequest("0", ImageType::Scene,false,true)};
		const std::vector<ImageResponse>& response = client.simGetImages(request);
		
		if (response.size() > 0) {
			ImageResponse image_info = response.at(response.size()-1);
			
			
			if (i == 0)
			{
				firstImageTime = image_info.time_stamp;
			}

			string imageTime = std::to_string((int)((image_info.time_stamp - firstImageTime)* 0.000001));

			std::string file_path = FileSystem::combine(path, to_string(i) /*+ "_"+ imageTime*/);
			std::ofstream file(file_path + ".bmp", std::ios::binary);
			file.write(reinterpret_cast<const char*>(image_info.image_data_uint8.data()), image_info.image_data_uint8.size());
			file.close();

			
			//video file
			videoFile << to_string(i) + " " + imageTime << endl;

			//////////////////////////////////////
			MultirotorState droneState = client.getMultirotorState();

			/*if (i == 0)
			{
				firstSensorTime = droneState.timestamp;
			}*/

			//string sateTime = std::to_string((int)((droneState.timestamp - firstSensorTime)* 0.000001));

			sensorFile << std::fixed << std::setprecision(7) << "GPS " + imageTime << " " << droneState.gps_location.latitude << " " << droneState.gps_location.longitude << " " << droneState.gps_location.altitude << " 0.0" << endl;

			

			float yawCamera, pitchCamera, rollCamera;
			VectorMath::toEulerianAngle(image_info.camera_orientation, pitchCamera, rollCamera, yawCamera);
			sensorFile << std::fixed << std::setprecision(7) << "ANGLE " << imageTime << " " << Utils::radiansToDegrees(yawCamera) << " " << Utils::radiansToDegrees(pitchCamera) << " " << Utils::radiansToDegrees(rollCamera) << " 0.0" << endl;
			cout << Utils::radiansToDegrees(yawCamera) << endl;

			/*float yawDrone, pitchDrone, rollDrone;
			VectorMath::toEulerianAngle(droneState.getOrientation(), pitchDrone, rollDrone, yawDrone);
			sensorFile << std::fixed << std::setprecision(7) << "ANGLE_DRONE " << sateTime << " " << Utils::radiansToDegrees(yaw) << " " << Utils::radiansToDegrees(pitch) + cameraPitch << " " << Utils::radiansToDegrees(roll) << " 0.0" << endl;

	*/

			VectorMath::Vector3f gyroAngularVel = VectorMath::toAngularVelocity(prevCameraOrientation, image_info.camera_orientation, (float)((image_info.time_stamp - prevImageTime)*0.000000001));

			//Vector3f gyroAngularVel = droneState.kinematics_estimated.twist.angular;

			sensorFile << std::fixed << std::setprecision(7) << "GYRO " << imageTime << " " << gyroAngularVel.x() << " " << gyroAngularVel.y() << " " << gyroAngularVel.z() << " 0.0" << endl;

			sensorFile << std::fixed << std::setprecision(7) << "PRESSURE " << imageTime << " " << droneState.gps_location.altitude << " 0.0 0.0 0.0" << endl;

			sensorFile << "GPS_SAT_NUMBER " << imageTime << " 6.000000 0.000000 0.000000 0.000000" << endl;

		
			prevCameraOrientation = image_info.camera_orientation;
			//prevDroneStateTime = droneState.timestamp;
			prevImageTime = image_info.time_stamp;

		}

		
		i++;
	}
}`

Hezzzzzy, 2019-04-29:
> write your own thread which will run and take the results according to the FPS you need CPP :
>
> `void recordImagesThread()
> {
> TTimePoint prevImageTime = 0;
> int i = 0;
> t1 = [...]

thx for ur answer, I already solve this problem by a stupid way, I slow down the clock speed....

closed 2019-05-05 by Hezzzzzy