microsoft/AirSim #1916: Reprojecting points between views

dfcnxngk, 2019-04-26:
I want to reproject 3d points acquired from depth map of the first view to another view. 
I need intrinsic and extrinsic camera parameters, and 3d points, reconstructed in the camera's coordinates. I didn't find all of the required functions in API. And is there a detailed documentation (with formulas) on projection model, which is used for rendering in the CV mode?

stavBodik, 2019-04-29:
please read here : 

https://github.com/unrealcv/unrealcv/issues/14

dfcnxngk, 2019-04-29:
@stavBodik Thank you! I've used that parametrization, but I'm missing something, because the result is not as expected to be. 
I used the Landscape Mountains environment, and got depth map and camera extrinsics by calling
```
resp = client.simGetImages([airsim.ImageRequest("2", airsim.ImageType.DepthPerspective,pixels_as_float=True)])
camera_info = client.simGetCameraInfo(str(0))
```
Changing the camera's position by 
```
camera_info.pose.position.z_val -= 1
client.simSetVehiclePose(airsim.Pose(camera_info.pose.position,
                                         airsim.to_quaternion(0, 0, 0)))
```
I noticed that optical axis doesn't coincide with Z axis in the world frame. I tried to test reprojection with camera translation only, so  camera-to-world matrix should be the following (considering axes swapping)
```
pos = {
'x_val': camera_info.pose.position.x_val,  
'y_val': camera_info.pose.position.y_val,  
'z_val': camera_info.pose.position.z_val 
}
MT = np.eye(4, dtype=np.float32)
MT[0:3, 3] = np.array( [pos['y_val'], pos['z_val'], pos['x_val']] )
```
In my case image has 320x240 pixels and FOV=90, so the focal length should be f=W/2, thus the intrinsic matrix will be
```
K = [ 
[160, 0, 160],
[0, 160, 120],
[0, 0, 1]
 ]

``` 
Then I reconstruct 3d points in the first camera coordinates
```
x = np.linspace(0, W-1, W)
y = np.linspace(0, H - 1, H)
x, y = np.meshgrid(x, y)
pts = np.ones((3, W * H), dtype=np.float32)
d = depth.flatten()
pts[0, :] = x.flatten()
pts[1, :] = y.flatten()
Kinv = np.linalg.inv(K)
ptsn = np.matmul(Kinv, pts)
Xn = np.reshape(ptsn[0, :], newshape=(H, W))
Yn = np.reshape(ptsn[1, :], newshape=(H, W))
pts3d = np.ones((3, W * H), dtype=np.float32)
# f=1 since we multiplied by Kinv
pts3d[0, :] = (d*Xn.flatten())
pts3d[1, :] = (d*Yn.flatten())
pts3d[2, :] = depth.flatten()
```
After that we should transfer points to the world coordinate frame
```
pts3dh = np.ones((4, pts3d_1.shape[1]))
pts3dh[0:3, :] = pts3d
pts3dh_1_to_world = np.matmul(M1, pts3dh)
```
And to the coordinate frame of the second camera
```
M2inv = np.linalg.inv(M2)
pts3dh_1_to_2 = np.matmul(M2inv, pts3dh_1_to_world )
```
Finally I reproject transfered points on the image plane
```
xn = (K[0,0]*pts3dh_1_to_2[0,:]/pts3dh_1_to_2[2,:]) + K[0,2]
yn = (K[1,1]*pts3dh_1_to_2[1,:]/pts3dh_1_to_2[2,:]) + K[1,2]
x_repr = np.reshape(xn, newshape=(H, W)).astype(np.int)
y_repr = np.reshape(yn, newshape=(H, W)).astype(np.int)
```
and render the reprojection image
```
for i in range(H):
    for j in range(W):
        if (xn[i,j] >= 0) and (xn[i,j] < W) and (yn[i,j] >= 0) and (yn[i,j] < H):
            img_out[y_repr[i,j], x_repr[i,j],:] = img1[i,j,:]
```
Here's what I've got:
- image_1
[image]
- image_2
[image]
- reprojection of image_1 on image_2
[image]
If we dont consider interpolation errors and border effects, the reprojection should be very similar to the image_2, but it's not. For example, as one can see, the shadows in the middle are not correctly distorted. Can't figure out what is missing here.

stavBodik, 2019-04-29:
@dfcnxngk

I don't really understand what you are trying to do, what do you mean :
"I want to reproject 3d points acquired from depth map of the first view to another view"

Trying to find the 3D coordinate of each pixel using 2 images by sending rays and checking where they are  intersect in space ?

Any way I see problem here :
"With camera translation only, so camera-to-world matrix should be the following"

please check if the camera to world matrix in both of your images has 0 rotation, are you sure the yaw,pitch and roll of the camera are 0,0,0 in both of images?

CameraMatrix = IntrinsicMatrix * ExtrinsicMatrix

CameraMatrix = ProjectionMatrix projMatrix = client.simGetCameraInfo("0").proj_mat;

edgarriba, 2019-04-30:
@dfcnxngk consider using [cv2.remap](https://docs.opencv.org/master/da/d54/group__imgproc__transform.html#gab75ef31ce5cdfb5c44b6da5f3b908ea4) to avoid bugs in you nested loop for rendering. Besides, in case you want to work on the GPU I have implemented a differentiable version in pytorch for warping by the depth here: https://torchgeometry.readthedocs.io/en/latest/warping.html#torchgeometry.DepthWarper.forward

dfcnxngk, 2019-04-30:
@stavBodik 
>I don't really understand what you are trying to do, what do you mean :
>"I want to reproject 3d points acquired from depth map of the first view to another view"

I need ground truth for training my model for image matching, and I want to check consistency of relative camera parameters and depth maps. If we have depth map and known camera parameters, we can reconstruct 3d points in the camera's coordinates frame. I'm trying to project theese points on the image plane of the second camera. If everything is correct, then correspondent points should concide, i.e. the new image (or a part of it, w.r.t. projective distortion) should be the same as the image of the second view.

>please check if the camera to world matrix in both of your images has 0 rotation, are you sure the yaw,pitch and roll of the camera are 0,0,0 in both of images?

Yes, it must be zero. I initialize pose with zero rotation:
```
camera_info = client.simGetCameraInfo(str(0))
client.simSetVehiclePose(airsim.Pose(camera_info.pose.position,
                                         airsim.to_quaternion(0, 0, 0)))
```
Then at each iteration I modify camera:
```
camera_info.pose.position.x_val += 1
camera_info.pose.position.y_val += 1
camera_info.pose.position.z_val -= 1
client.simSetVehiclePose(airsim.Pose(camera_info.pose.position,
                                         airsim.to_quaternion(0, 0, 0)))
```
Is it normal that at each iteration ```camera_info = client.simGetCameraInfo(str(0))``` returns the same projection matrix?
```
[[-4.37113918e-08  0.00000000e+00  0.00000000e+00  0.00000000e+00]
 [ 0.00000000e+00 -5.82818558e-08  0.00000000e+00  0.00000000e+00]
 [ 0.00000000e+00  0.00000000e+00  0.00000000e+00  1.00000000e+00]
 [ 0.00000000e+00  0.00000000e+00  1.00000000e+01  0.00000000e+00]]
```
If P = K[ R | t], then with identity rotation and FOV=90 there shoud be values W/2=160 on the main diagonal? But here we can see some unknown additional scaling.

@edgarriba
Thanks a lot for the provided link on torchgeometry! That's a great tool!

stavBodik, 2019-04-30:
@dfcnxngk 

"Is it normal that at each iteration camera_info = client.simGetCameraInfo(str(0)) returns the same projection matrix?"

Nope I so it also when tryd to reconstruct the intrinsic matrix from it.

About the 0,0,0 rotation, please check that the rotation is really 0, you can set it to 0 but maybe they have bug because there is wind effects etc that airsim applying in order to demonstraate movement noise.

dfcnxngk, 2019-04-30:
I used python
```
(pitch, roll, yaw) = airsim.to_eularian_angles(client.simGetCameraInfo("0").pose.orientation)
```
and angles are always zero. 

But I noticed that different responses have different camera position, while orientation is the same. The following code
```
responses = client.simGetImages([
        airsim.ImageRequest("0", airsim.ImageType.Scene),
        airsim.ImageRequest("1", airsim.ImageType.DepthVis),
        airsim.ImageRequest("2", airsim.ImageType.DepthPerspective)])

for i, response in enumerate(responses):
       if response.pixels_as_float:
            print("Type %d, size %d, pos %s, orient %s" % (response.image_type, 
            len(response.image_data_float),
            pprint.pformat(response.camera_position),
            pprint.pformat(response.camera_orientation)))
           
        else:
            print("Type %d, size %d, pos %s, orient %s" % (response.image_type, 
            len(response.image_data_uint8),
            pprint.pformat(response.camera_position),
              pprint.pformat(response.camera_orientation)))
```
gives
```
# iteration 0

Type 0, size 115349, pos <Vector3r> {   'x_val': 0.0,
    'y_val': 0.0,
    'z_val': -0.0}, orient <Quaternionr> {   'w_val': 1.0,
    'x_val': -0.0,
    'y_val': -0.0,
    'z_val': 0.0}
Type 3, size 14802, pos <Vector3r> {   'x_val': 0.0,
    'y_val': 0.125,
    'z_val': -0.0}, orient <Quaternionr> {   'w_val': 1.0,
    'x_val': -0.0,
    'y_val': -0.0,
    'z_val': 0.0}
Type 2, size 76800, pos <Vector3r> {   'x_val': 0.0,
    'y_val': -0.125,
    'z_val': -0.0}, orient <Quaternionr> {   'w_val': 1.0,
    'x_val': -0.0,
    'y_val': -0.0,
    'z_val': 0.0}

#iteration 1

Type 0, size 116223, pos <Vector3r> {   'x_val': 1.0,
    'y_val': 1.0,
    'z_val': -1.0}, orient <Quaternionr> {   'w_val': 1.0,
    'x_val': -0.0,
    'y_val': -0.0,
    'z_val': 0.0}
Type 3, size 14820, pos <Vector3r> {   'x_val': 1.0,
    'y_val': 1.125,
    'z_val': -1.0}, orient <Quaternionr> {   'w_val': 1.0,
    'x_val': -0.0,
    'y_val': -0.0,
    'z_val': 0.0}
Type 2, size 76800, pos <Vector3r> {   'x_val': 1.0,
    'y_val': 0.875,
    'z_val': -1.0}, orient <Quaternionr> {   'w_val': 1.0,
    'x_val': -0.0,
    'y_val': -0.0,
    'z_val': 0.0}

#iteration 2

Type 0, size 117678, pos <Vector3r> {   'x_val': 2.0,
    'y_val': 2.0,
    'z_val': -2.0}, orient <Quaternionr> {   'w_val': 1.0,
    'x_val': -0.0,
    'y_val': -0.0,
    'z_val': 0.0}
Type 3, size 14893, pos <Vector3r> {   'x_val': 2.0,
    'y_val': 2.125,
    'z_val': -2.0}, orient <Quaternionr> {   'w_val': 1.0,
    'x_val': -0.0,
    'y_val': -0.0,
    'z_val': 0.0}
Type 2, size 76800, pos <Vector3r> {   'x_val': 2.0,
    'y_val': 1.875,
    'z_val': -2.0}, orient <Quaternionr> {   'w_val': 1.0,
    'x_val': -0.0,
    'y_val': -0.0,
    'z_val': 0.0}
```
So the depth camera has shift by Y axis. Will try to consider that.

dfcnxngk, 2019-05-06:
Converting perspective depth to the plane depth using this

https://gist.github.com/edz-o/84d63fec2fc2d70721e775337c07e9c9

seems to solve my problem

closed 2019-05-06 by dfcnxngk

Zzerr0r, 2022-04-28:
@AlexanderZ3 have you been able to solve this? I am currently stuck with the same problem of not knowing how to use the camerInfo().proj_mat