microsoft/AirSim #1917: Semantic labels of Point Cloud

vishal-prgmr, 2019-04-26:
Hi Community,

I am interested in associating the point cloud with the semantic labels. I was digging into the UnrealLidarSensor.hpp to see if I can tap out the associated actor from the ray tracing function.

Has anyone else done this? Any heads up ? 

Thank You

vishal-prgmr, 2019-05-05:
Will submit a PR now for you guys to review it.

closed 2019-05-05 by vishal-prgmr

CharlieRuan0130, 2021-04-27:
> Will submit a PR now for you guys to review it.

> Hi Community,
> 
> I am interested in associating the point cloud with the semantic labels. I was digging into the UnrealLidarSensor.hpp to see if I can tap out the associated actor from the ray tracing function.
> 
> Has anyone else done this? Any heads up ?
> 
> Thank You

Hi v-prgmr, I am also looking into giving point cloud data labels. I am wondering if you have achieved it. If so, could you provide some insights on how to do that? Thank you!