microsoft/AirSim #1919: Image Resolution under CaptureSettings

vishal-prgmr, 2019-04-26:
Hi Folks,

I have setup my settings.json as follows. I have also read the other threads about not being able to get the set resolution when using the simGetImages APIs. 

My settings.json is as follows. I am not able figure out the mistake. Let me know what I am doing wrong. 

Thank You

`{

    "SettingsVersion": 1.2,

    "SimMode": "Car",
	"RecordUIVisible": true,
	"LogMessagesVisible": true,
	"ViewMode": "",
	"Recording": {
    "RecordOnMove": false,
    "RecordInterval": 0.05,
    "Cameras": [
        { "CameraName": "front_center", "ImageType": 0, "PixelsAsFloat": false, "Compress": true },
		{ "CameraName": "front_center", "ImageType": 3, "PixelsAsFloat": false, "Compress": true }
    ]
	
	},
	"CameraDefaults": { 
	"CaptureSettings": [
      {
        "ImageType": 0,
        "Width": 1280,
        "Height": 960,
        "FOV_Degrees": 90
       },
	   {
        "ImageType": 3,
        "Width": 1280,
        "Height": 960,
        "FOV_Degrees": 90
       },
	   {
        "ImageType": 5,
        "Width": 1280,
        "Height": 960,
        "FOV_Degrees": 90
       }
	   
    ]
	},
    
	"SubWindows": [
					{"WindowID": 0, "ImageType": 0, "CameraName": "front_center", "Visible": true},
					{"WindowID": 1, "ImageType": 3, "CameraName": "front_center", "Visible": true},
					{"WindowID": 2, "ImageType": 5, "CameraName": "front_center", "Visible": true}
				  ],
     "Vehicles": {
        "EgoCar": {
            "VehicleType": "PhysXCar",
            "AutoCreate": true,
			"Cameras": {
				"front_center": {
				"CaptureSettings": [
				  {
					"ImageType": 0,
					"width": 1280,
					"Height": 960,
					"FOV_Degrees": 90
					
				  },
				  {
					"ImageType": 3,
					"width": 1280,
					"Height": 960,
					"FOV_Degrees": 90
					
				  },
				  {
					"ImageType": 5,
					"width": 1280,
					"Height": 960,
					"FOV_Degrees": 90
					
				  }
				],
					
				
				"X": 0, "Y": 0, "Z": -1.9,
				"Pitch": 0, "Roll": 0, "Yaw": 0    
				}
			  },
			
			"Sensors": {
                "LidarSensor1": { 
                    "SensorType": 6,
                    "Enabled" : true,
                    "NumberOfChannels": 16,
					"Range": 100,
                    "RotationsPerSecond": 10,
                    "PointsPerSecond": 100000,
                    "X": 0, "Y": 0, "Z": -1,
                    "Roll": 0, "Pitch": 0, "Yaw" : 0,
                    "VerticalFOVUpper": 10,
                    "VerticalFOVLower": -10,
                    "HorizontalFOVStart": -360,
                    "HorizontalFOVEnd": 360,
                    "DrawDebugPoints": true,
                    "DataFrame": "VehicleInertialFrame"
                }		
            }
        }
    }
}`

Hezzzzzy, 2019-04-29:
if u recorded the image by pressing R, u only need change the 
"CaptureSettings": [
  {
    "Width": 1280,
    "Height": 960
   }

the others is not necessary

vishal-prgmr, 2019-04-29:
@Hezzzzzy , this is my API call,

-----------
   responses = self.client.simGetImages([airsim.ImageRequest("front_center", airsim.ImageType.Scene, False, False)])
        for response in responses:
            img1d = numpy.fromstring(response.image_data_uint8, dtype=numpy.uint8)
            img_rgba = img1d.reshape(response.height, response.width, 4)
            img_rgba = numpy.flipud(img_rgba)
-----------
And i have also set the resolution for the **scene** image type in the settings.json (you can see above)

Only the height is getting set according to the settings.json and the width is still 254. 

I am out of ideas to solve this. I have read all the issues related to this too.

Hezzzzzy, 2019-04-29:
obviously,  the error is the way u using the api parameter,
self.client.simGetImages([airsim.ImageRequest([image type], airsim.ImageType.Scene, False, False)])

can u understand?  u only change the image type 0, 3 , 5 the image width and height, but in ur code, u get the image type is "front_center", u did not change the "front_center" image type setting

vishal-prgmr, 2019-04-29:
@Hezzzzzy , Clear.
My bad that I assumed the first parameter of the simgetImages to be the camera name from which the images are captured.

Thank You :)

vishal-prgmr, 2019-04-29:
> obviously, the error is the way u using the api parameter,
> self.client.simGetImages([airsim.ImageRequest([image type], airsim.ImageType.Scene, False, False)])
>
> can u understand? u only change the [...]

@Hezzzzzy , the first parameter in the api call is not image type it is the camera from which you want to grab the images.

vishal-prgmr, 2019-04-29:
@msb336 , I have created a custom camera at a desired location, and in the sub windows the resolution is the same as I want it to be. However when i call the simGetImages API with the camera name as my custom camera, only the height of the captured image follows my specification (given in the .json file). 

How do i go about this?

Hezzzzzy, 2019-04-29:
@vishal-prgmr 
hi bro, I had made a mistake, hope u can forgive me,u are right, the first para is camera name,
according to ur setting file, u created a new camera name "front-center", u want to get this camera's images, ok?
I tried to made a test, it's no problem, I will show u my code, maybe it will help u

there is my setting file:
`{
  "SeeDocsAt": "https://github.com/Microsoft/AirSim/blob/master/docs/settings.md",
  "SettingsVersion": 1.2,
  "CameraDefaults": {
    "CaptureSettings": [
      {
        "ImageType": 0,
        "Width": 1280,
        "Height": 800

      }
    ]
  },

  "Recording": {
    "RecordInterval": 1
  },
  "ClockSpeed": 1,
  "Vehicles": {
    "car1": {
      "VehicleType": "PhysXCar",
      "Cameras": {
        "front-center": {
          "CaptureSettings": [
            {
              "ImageType": 0,
              "Width": 1280,
              "Height": 800,
              "FOV_Degrees": 90
            }
          ],
          "X": 0,
          "Y": 0,
          "Z": -1.9,
          "Pitch": 0,
          "Roll": 0,
          "Yaw": 0
        }
      }
    }
  }
  }
`
there is my test python code:
`
client = airsim.CarClient()
num = 0

while True:
    num += 1
    time.sleep(3)
    a = client.simGetImages([airsim.ImageRequest("front-center", airsim.ImageType.Scene, False, True)])
    for item in a:
        print('width: {}'.format(item.width))
        print('height: {}'.format(item.height))
`

vishal-prgmr, 2019-04-29:
@Hezzzzzy , Hey thanks for the reply. Well, nothing changed.  Your API call is the same as mine and the image resolution is still the same. The height is 960 but the width is still 254.

madratman, 2019-04-29:
@vishal-prgmr, I finally figured out after a full 15 minutes of being [dazed and confused](https://www.youtube.com/watch?v=-rmtJ6-EY1E)
This is happening as you have a type in your settings.json. 
It's saying `width` instead of `Width`. scrip
Yes, as the w is not capitol, the JSON parser is not reading that argument behind the scenes and defaulting to the default width, which is 254. 

Also, your settings.json is defining the custom camera twice, which is unnecessary. 
Finally, there's another potentially another source of error in @Hezzzzzy's python calls. 
They're using `front-center` and not `front_center`. Be careful with that. 
For completeness. The default camera is also called `front-center` FYI, you are making a new cam called `front_center`

Here's the settings.json you need:

```JSON
{
  "SettingsVersion": 1.2,

  "SimMode": "Car",
  "RecordUIVisible": true,
  "LogMessagesVisible": true,
  "ViewMode": "",
  "Recording": {
  "RecordOnMove": false,
  "RecordInterval": 0.05,
  "Cameras": [
      { "CameraName": "front_center", "ImageType": 0, "PixelsAsFloat": false, "Compress": true },
      { "CameraName": "front_center", "ImageType": 3, "PixelsAsFloat": false, "Compress": true }
    ]
  
  },
    
  "SubWindows": [
          {"WindowID": 0, "ImageType": 0, "CameraName": "front_center", "Visible": true},
          {"WindowID": 1, "ImageType": 3, "CameraName": "front_center", "Visible": true},
          {"WindowID": 2, "ImageType": 5, "CameraName": "front_center", "Visible": true}
          ],
     "Vehicles": {
        "EgoCar": {
            "VehicleType": "PhysXCar",
            "AutoCreate": true,
      "Cameras": {
        "front_center": {
        "CaptureSettings": [
          {
          "ImageType": 0,
          "Width": 1280,
          "Height": 960,
          "FOV_Degrees": 90
          
          },
          {
          "ImageType": 3,
          "Width": 1280,
          "Height": 960,
          "FOV_Degrees": 90
          
          },
          {
          "ImageType": 5,
          "Width": 1280,
          "Height": 960,
          "FOV_Degrees": 90
          
          }
        ],
          
        
        "X": 0, "Y": 0, "Z": -1.9,
        "Pitch": 0, "Roll": 0, "Yaw": 0    
        }
        },
      
      "Sensors": {
                "LidarSensor1": { 
                    "SensorType": 6,
                    "Enabled" : true,
                    "NumberOfChannels": 16,
                    "Range": 100,
                    "RotationsPerSecond": 10,
                    "PointsPerSecond": 100000,
                    "X": 0, "Y": 0, "Z": -1,
                    "Roll": 0, "Pitch": 0, "Yaw" : 0,
                    "VerticalFOVUpper": 10,
                    "VerticalFOVLower": -10,
                    "HorizontalFOVStart": -360,
                    "HorizontalFOVEnd": 360,
                    "DrawDebugPoints": true,
                    "DataFrame": "VehicleInertialFrame"
                }   
            }
        }
    }
}
```

vishal-prgmr, 2019-04-29:
Omg! I see that now! I didn't check the case at all. Yeah I defined the camera twice because I wanted to see if explicitly mentioning the resolution for the specific camera would do something. 

Thank you so much for your time and reply!

closed 2019-04-29 by vishal-prgmr