microsoft/AirSim #1923: UE 4.18 reduces quality of the mesh!! why? how to manage it?

Supaple-x, 2019-04-29:
Hi guys, 
I fly a lot. and after drone survey and successfully built model (beautiful and high qualified) I try to put a map into AirSim and...
UE wastes all my work! It makes mesh simple. And unbelievable ugly :(
Is there way to mange this?

thanks

RVBldr, 2019-04-30:
It would be great to see some screen shots as well as what that environment looks like in the UE editor.

msb336, 2019-06-04:
Unreal might be [auto-LODing](https://docs.unrealengine.com/en-US/Engine/Content/Types/StaticMeshes/HowTo/AutomaticLODGeneration/index.html) your mesh. This issue seems isolated to the Unreal Engine, so you may want to post your issue on [unreal's forums](https://forums.unrealengine.com/)

closed 2019-06-04 by msb336

Supaple-x, 2019-07-18:
Thanks guys! I've found auto-LODing. It seems that I had not enough resources on my notebook.
I bought powerful station and now it works perfect!