microsoft/AirSim #1924: LIDAR causes external camera to darken extremely

jaymefosa, 2019-04-30:
External view with LIDAR on:

[image]

External view with LIDAR off:

[image]

Unsure if this is an Unreal shader issue, there were no problems on the hello world Landscape map.

lcyawym, 2019-04-30:
Can you show your setting file. I can't use lidar ，It always occurs  no vehicle named “Drone1”

RVBldr, 2019-04-30:
Seems strange. Yes, how about posting the settings file?

lcyawym, 2019-04-30:
{
  "SeeDocsAt": "https://github.com/Microsoft/AirSim/blob/master/docs/settings_json.md",
  "SettingsVersion": 1.2,

  "SimMode": "Multirotor",
  "Vehicles": {
    "Drone1": {
      "VehicleType": "SimpleFlight",
      "DefaultVehicleState": "Armed",
      "AutoCreate": true,
      "PawnPath": "",
      "EnableCollisionPassthrogh": false,
      "EnableCollisions": true,
      "AllowAPIAlways": true,
      "Sensors": {
        "MyLidar1": { 
            "SensorType": 6,
            "Enabled" : true,
            "NumberOfChannels": 16,
            "PointsPerSecond": 10000,
            "X": 0, "Y": 0, "Z": -1,
            "DrawDebugPoints": true
        }
    }
  },
  "CameraDefaults": {
    "CaptureSettings": [
      {
        "ImageType": 3,
        "Width": 64,
        "Height":64,
        "FOV_Degrees": 90,
        "AutoExposureSpeed": 100,
        "AutoExposureBias": 0,
        "AutoExposureMaxBrightness": 0.64,
        "AutoExposureMinBrightness": 0.03,
        "MotionBlurAmount": 0,
        "TargetGamma": 1.0,
        "ProjectionMode": "",
        "OrthoWidth": 5.12
      }
    ]
  }
}
}
Whether I need to update the plugins in my project?

jaymefosa, 2019-05-01:
Here are the settings that produced the darkened results above:

```
{
	"SeeDocsAt": "https://github.com/Microsoft/AirSim/blob/master/docs/settings.md",
	"SettingsVersion": 1.2,
	"SimMode": "Multirotor",

     "Vehicles": {
		"Drone1": {
			"VehicleType": "simpleflight",
			"AutoCreate": true,
			"Sensors": {
			    "LidarSensor1": { 
					"SensorType": 6,
					"Enabled" : false,
					"NumberOfChannels": 16,
					"RotationsPerSecond": 10,
					"PointsPerSecond": 100000,
					"X": 0, "Y": 0, "Z": -1,
					"Roll": 0, "Pitch": 0, "Yaw" : 0,
					"VerticalFOVUpper": -15,
					"VerticalFOVLower": -25,
					"HorizontalFOVStart": -20,
					"HorizontalFOVEnd": 20,
					"DrawDebugPoints": true,
					"DataFrame": "VehicleInertialFrame"
				},
				"LidarSensor2": { 
				   "SensorType": 6,
					"Enabled" : false,
					"NumberOfChannels": 4,
					"RotationsPerSecond": 10,
					"PointsPerSecond": 10000,
					"X": 0, "Y": 0, "Z": -1,
					"Roll": 0, "Pitch": 0, "Yaw" : 0,
					"VerticalFOVUpper": -15,
					"VerticalFOVLower": -25,
					"DrawDebugPoints": true,
					"DataFrame": "VehicleInertialFrame"
				}
			}
		}
    }
}
```

lcyawym, 2019-05-01:
@jaymefosa maybe the "Enable“ of sensors is should be true?

jaymefosa, 2019-05-01:
@lycawym haha right, sorry, this is the version used to create the second image I posted above. If the sensors are enabled, it causes the high contrast darkening issue

lcyawym, 2019-05-01:
@jaymefosa haha，thanks for your reply，I want to use lidar to get  point cloud message. What I did  is to use block project,and change the setting files as above. and when I use getLidarData(), It occurs that "RPCError: rpclib: function 'getLidarData' (called with 2 arg(s)) threw an exception. The exception contained this information: No lidar with name LidarSensor1 exist on vehicle.`"
Do you know what wrong with my step?thanks

jaymefosa, 2019-05-01:
@lcyawym Can you post your getLidarData() code?

In your settings it looks like the name of the sensor is:

```
  "Sensors": {
        "MyLidar1": {
```

lcyawym, 2019-05-01:
lidar_data = self.client.getLidarData()
def parse_lidarData(self, data):
		# reshape array of floats to array of [X,Y,Z]
		points = np.array(data.point_cloud, dtype=np.dtype('f4'))
		points = np.reshape(points, (int(points.shape[0]/3), 3))
		return points	
I also change to
lidar_data = self.client.getLidarData(lidar_name = 'MyLidar1', vehicle_name = 'Drone1')
but it doesn't work

jaymefosa, 2019-05-01:
Hmmmmm.. so, try copying my settings.. Did you try that yet?

For example the problem could be the 'yaw roll pitch' not being in your settings. Someone else pointed out if they are missing then it caused problems and I also had a similar issue.

lcyawym, 2019-05-01:
File "/home/liuchunyuan/rl/AirSim/3D_path_finding/DDPG/drone_env.py", line 214, in step
    lidar_data = self.client.getLidarData(lidar_name = 'LidarSensor1', vehicle_name = 'Drone1')
  File "/home/liuchunyuan/anaconda2/envs/ddpgair/lib/python3.5/site-packages/airsim/client.py", line 135, in getLidarData
    return LidarData.from_msgpack(self.client.call('getLidarData', lidar_name, vehicle_name))
  File "/home/liuchunyuan/anaconda2/envs/ddpgair/lib/python3.5/site-packages/msgpackrpc/session.py", line 41, in call
    return self.send_request(method, args).get()
  File "/home/liuchunyuan/anaconda2/envs/ddpgair/lib/python3.5/site-packages/msgpackrpc/future.py", line 45, in get
    raise error.RPCError(self._error)
msgpackrpc.error.RPCError: rpclib: function 'getLidarData' (called with 2 arg(s)) threw an exception. The exception contained this information: Vehicle API for 'Drone1' is not available. This could either because this is simulation-only API or this vehicle does not exist.
Also something wrong  - -

jaymefosa, 2019-05-01:
Could you post your complete settings.json again? 

And did you follow every step in this guide?: https://www.youtube.com/watch?v=1oY8Qu5maQQ

msb336, 2019-05-01:
@jaymefosa 
Does this issue appear if you use the same settings file in a different environment?
Since the LiDAR is working, my hunch is that there are problems with the interaction between the LiDAR debug points and your environment.

jaymefosa, 2019-05-01:
@msb336 
Yeah it works fine in the other environment (Landscape). It seems like some kind of gamma correction issue, turning debug points off causes the camera to slowly adjust to the lack of strong contrast, or something

lcyawym, 2019-05-02:
@jaymefosa thankyou，I solved my problem. And the point cloud I get use "VehicleInertialFrame"mode, if I want to calculate the distance from my vehicle to point cloud.If I need coordinate transformation? Or just calculate the distance of two point

jaymefosa, 2019-05-02:
@lcyawym
Great to hear! What was the issue in the end?

For the LIDAR setting, did you find it yet? If you use:  "SensorLocalFrame" instead of "VehicleInertialFrame" then you don't have to calculate the transform yourself

lcyawym, 2019-05-02:
yes I have found it,and it working.Now I need to calculate the distance from my vehicle to obstacles.So I need to change the mode to "SensorLocalFrame".And use like below can get the distance,right?
for point in point_cloud:
	point_dis = np.sqrt(point[0]**2 + point[1]**2+ point[2]**2)

jaymefosa, 2019-05-03:
@lcyawym yeah but I would make it a vector operation instead of _for each:_ point. _For each_ is really slow compared to numpy's vectorization

lcyawym, 2019-05-03:
Ok, I will try again. Thanks for answer me so many questions. haha

lcyawym, 2019-05-03:
@jaymefosa，hmmmm ,when I train my mode, I will need to use api "client.reset()",but this api will sometimes make the platform crash.Did you know how to solve ti? Sometimes when the reset() is working,the platform will close by itself.

jaymefosa, 2019-05-04:
@lcyawym ah sorry, I haven't gotten to training yet..

msb336, 2019-05-14:
Closing this issue, as it looks to be sourced in unreal. please re-open if you are still having problems.

closed 2019-05-14 by msb336