microsoft/AirSim #1928: Any way to run multiple PX4 drones in one environment?

KeyneZzz, 2019-05-05:
I tried to add a PX4 drone in my environment by editing the settings to follow:

```
  "Vehicles": {
    "PX4-3": {
      "VehicleType": "PX4Multirotor",
      "LogViewerHostIp": "127.0.0.1",
      "LogViewerPort": 14388,
      "OffboardCompID": 1,
      "OffboardSysID": 134,
      "QgcHostIp": "127.0.0.1",
      "QgcPort": 14550,
      "SerialBaudRate": 115200,
      "SerialPort": "COM3",
      "SimCompID": 42,
      "SimSysID": 142,
      "SitlIp": "127.0.0.1",
      "SitlPort": 14556,
      "UdpIp": "127.0.0.1",
      "UdpPort": 14560,
      "UseSerial": true,
      "VehicleCompID": 1,
      "VehicleSysID": 135,
      "Model": "Generic",
      "LocalHostIp": "127.0.0.1",
      "Model": "Hexacopter",
      "X": 0, "Y": 0, "Z": -3,
      "Pitch": 0,"Roll": 0,"Yaw": 0    
    },
    "PX4-13": {
      "VehicleType": "PX4Multirotor",
      "LogViewerHostIp": "127.0.0.1",
      "LogViewerPort": 14389,
      "OffboardCompID": 1,
      "OffboardSysID": 135,
      "QgcHostIp": "127.0.0.1",
      "QgcPort": 14551,
      "SerialBaudRate": 115200,
      "SerialPort": "COM13",
      "SimCompID": 42,
      "SimSysID": 143,
      "SitlIp": "127.0.0.1",
      "SitlPort": 14557,
      "UdpIp": "127.0.0.1",
      "UdpPort": 14561,
      "UseSerial": true,
      "VehicleCompID": 1,
      "VehicleSysID": 136,
      "Model": "Generic",
      "LocalHostIp": "127.0.0.1",
      "Model": "Hexacopter",
      "X": 4, "Y": 0, "Z": -3,
      "Pitch": 0,"Roll": 0,"Yaw": 0    
    }
  }
```
but it works like this: 
[image]

 I tried to control them by rc, but only the one on COM3 works. Any way to make it work? Thx

Dan-Moss, 2019-07-17:
Having the same issue

jonyMarino, 2021-09-01:
Now we have a tutorial for this.
Closed due to age.
Please feel free to open a new issue (referring to this one) if you still have this problem.
Thanks!

closed 2021-09-01 by jonyMarino