microsoft/AirSim #1930: BSCMAKE Error BK1509

Marxlp, 2019-05-06:
The error is shown below:
[image]

Does anyone have idea for this?

Marxlp, 2019-05-06:
I searched on Google, and I found this [image], which said 

> To fix by using the following possible solutions
> Free some disk space.
> 
> Increase the size of the swap file.
> 
> Increase the size of the Windows swap file.
> 
> Reduce the memory BSCMAKE requires by using /Ei or /Es to eliminate some input files or /Em to eliminate macro bodies.

But I are not famaliar with msbuild, so I still don't how to fix this problem.

msb336, 2019-10-10:
closing due to inactivity. please re-open if this is a continuing issue

closed 2019-10-10 by msb336