microsoft/AirSim #1932: getLidarData() - pose attribute

GuiBo79, 2019-05-06:
When I use the client.getLidarData() I receive an AttributeError as folow:

AttributeError: 'LidarData' object has no attribute 'pose'

Can be something wrong in settings.json ?

msb336, 2019-05-07:
@GuiBo79

If you are using the global version of the airsim python module
i.e. you did something like this:
```
pip install python
```
It is not at it's most up to date version, so there is no pose attribute in the LidarData object stored in types.py.

You must use the local module located in AirSim/PythonClient. You will see some examples of how to import the local airsim module [here](https://github.com/microsoft/AirSim/tree/master/PythonClient) as well.

madratman, 2019-05-14:
We have updated the pypi package here https://pypi.org/project/airsim/ and a `$ pip install airsim --upgrade --user`, or `$ sudo pip install airsim --upgrade` should do the trick

+1 @msb336 
@GuiBo79 in general, you can do a 
```
cd PythonClient;
python setup.py install  # use sudo or if you get a permissions error.  
```
to install it locally as mentioned [here](https://github.com/microsoft/AirSim/issues/1624#issuecomment-487815855)

closed 2019-05-14 by madratman