microsoft/AirSim #1940: Can I change the UAV model?

makeisbad, 2019-05-07:
# Read This First

## If you are reporting a bug
* Make sure to write **all reproduction steps**
* Include full error message in text form
* Search issues for error message before filing issue
* Attach screenshot if applicable
* Include code to run if applicable

## If you have question
* Add clear and concise title
* Add OS, AirSim version, Python version, Unreal version if applicable
* Include context on what you are trying to achieve
* Include details of what you already did to find answers

**What's better than filing issue? Filing a pull request :).**

------------------------------------ (Remove above before filing the issue) ------------------------------------

eiriksfa, 2019-05-07:
Check out this tutorial/guide: https://github.com/microsoft/AirSim/wiki/hexacopter

makeisbad, 2019-05-07:
> Check out this tutorial/guide: https://github.com/microsoft/AirSim/wiki/hexacopter

Thank you very much.I still have a problem.Can I use my own model?My model is an unmanned airship.

jonyMarino, 2020-10-09:
closed due to inactivity. Please feel free to reopen for your last question.

closed 2020-10-09 by jonyMarino