microsoft/AirSim #1942: The imu data is not accurate

Hezzzzzy, 2019-05-07:
hi folks,
@madratman     I get the imu data by airsim.CarClient().getImuData(),
I recoded a piece of data about acrossing the curve road, then I fit the trajectory by using median integration, compare with the GPS trajectory, has something different

this is the image of the GPS trajectory and imu median integeration trajectory：
[https://photos.google.com/photo/AF1QipMYcS9xsDqILXNt3ZmE2f_cM4jDAnfeqaTFyP8p](url)

kylelindgren, 2019-05-07:
Link appears to be broken

Hezzzzzy, 2019-05-08:
> Link appears to be broken

I have tried it, good running...
there are two lines in the image , one GPS trajectory,  the other one is imu trajectory, the imu trajectory has obvious offset

Hezzzzzy, 2019-05-13:
@kylelindgren  ahaha, this problem has been solved  by an alternative method, get the imu data by speed scalar and orientation, I have checked it , it's no problem

jonyMarino, 2020-11-23:
Closed due to age.
Please feel free to open a new issue (referring to this one) if you still have this problem, and include the following (as requested when opening a bug issue)

*Make sure to write all reproduction steps
*Include full error message in text form
*Search issues for error message before filing an issue
*Attach a screenshot if applicable
*Include code to run if applicable

Thanks!

closed 2020-11-23 by jonyMarino