microsoft/AirSim #1943: Data recorded mismatch real ones

San-Holo, 2019-05-07:
Hello everyone and thanks in advance for your help,

I'm recording several infos by myself using the API, so without using the "r" record button in the simulation. Everything's fine with this process, but my problem concerns the gathered data. 

In fact, the trajectory is well recorded, but it's just like a translation was applied to it. So for a straight line, if I use simSetVehiclePose() for each line of my csv file, one with the right orientation is observed, but the car is literally far from where I started to record.

I tried to find related issues, but as far as my knowledge goes, I couldn't find at least one.

Concerning the code, I'm just using simGetVehiclePose() at a specified timestep to gather data, and nothing more than the opposite function when "replaying".

Do you have any ideas about it ?

msb336, 2019-05-14:
@San-Holo Let me make sure I am understanding you correctly:

You are attempting to record a car's position using `simGetVehiclePose()` while it is driving along some path that you define, and then replay that trajectory using `simSetObjectPose()` later. Your issue is that `simSetVehiclePose()` is placing your object at a different location in the scene?

If this is the case, are you initializing your drone in the same pose/position every time? If not, the origin in your world will be different between instances, causing a linear offset.

Can you post a video of the phenomenon to this thread?

msb336, 2019-06-06:
Closing due to lack of activity. please re-open if you are still having problems related to this issue

closed 2019-06-06 by msb336