microsoft/AirSim #1946: Editing pre-built environments for my own project

eds89, 2019-05-09:
Hello. I'd like to edit the Neighbourhood environment for my own custom project. Where can I find its source code? Should I try to reverse-engineer it from the binary?

msb336, 2019-05-09:
@eds89 Unfortunately, the Neighborhood environment contains non open-source content, so we can't make the source code available to our users. You are more than welcome to construct your own custom environments (and share the binaries with all your pals at AirSim!)

closed 2019-05-09 by msb336

eds89, 2019-05-09:
How much would it cost to acquire a license for the neighborhood
environment? By the way, is there any environment available with exception
of the 'Blocks' environment?

msb336, 2019-05-09:
@eds89 we do not distribute licenses our environments. Blocks is the only environment users have full access to.

If you are looking to add noise to sensor data or simulate an adversarial attack, doing so to the sensor data after it has been received by the client will be your path of least resistance.

eds89, 2019-05-10:
Hi. If I change the sensor data after it has been received by the client,
how can I assess how the DNN behaves. The way I modelled the adversarial
attacks are through computer-graphics effects (which are implemented as
post-processing materials in Unreal).