microsoft/AirSim #1949: Retry connection over the limit

cunyu008, 2019-05-09:
when i run hellocar.py ,I got some errors...
WARNING:tornado.general:Connect error on fd 468: WSAECONNREFUSED
WARNING:tornado.general:Connect error on fd 1036: WSAECONNREFUSED
WARNING:tornado.general:Connect error on fd 1036: WSAECONNREFUSED
WARNING:tornado.general:Connect error on fd 804: WSAECONNREFUSED
WARNING:tornado.general:Connect error on fd 804: WSAECONNREFUSED
Traceback (most recent call last):
  File "E:\AirSim\PythonClient\car\hello_car.py", line 10, in <module>
    client.confirmConnection()
  File "E:\AirSim\PythonClient\airsim\client.py", line 54, in confirmConnection
    if self.ping():
  File "E:\AirSim\PythonClient\airsim\client.py", line 24, in ping
    return self.client.call('ping')
  File "E:\Anaconda\lib\site-packages\msgpackrpc\session.py", line 41, in call
    return self.send_request(method, args).get()
  File "E:\Anaconda\lib\site-packages\msgpackrpc\future.py", line 43, in get
    raise self._error
msgpackrpc.error.TransportError: Retry connection over the limit
i would be grateful if anyone help me!

msb336, 2019-05-09:
@cunyu008 this error is thrown when the airsim client can't find a server to connect to. What unreal environment do you have open when you run `hellocar.py`?

closed 2019-10-08 by msb336

zenghaogithub, 2020-03-31:
> @cunyu008 this error is thrown when the airsim client can't find a server to connect to. What unreal environment do you have open when you run `hellocar.py`?

how to install UE4 environment rightly so that the sirsim can run successfully？

ArthurYdalgo, 2021-07-10:
Not sure if it's the same thing that was happening to me, but check the port you're using to access the simulation. Depending on the environment you're using, the port might be 42451 (the neighbourhood, for example). The City and Blocks envs, on the other hand, the port is 41451 (it might also deppend on the version you're using, but I can't confirm that). I struggled with this for about 3 days until I realized this could be it, because I remember that somewhere I saw 2 different ports. 

I hope it helps, bye!

zimmy87, 2021-07-23:
Hi @ArthurYdalgo, which versions of the neighbourhood, City, and Blocks environments are you using? In the current master, the port used for RPC communication defaults to 41451. Unless you're using a settings file exclusive to a specific environment overriding the default value, you shouldn't see communication over a different port based on environment alone.

ArthurYdalgo, 2021-07-26:
> Hi @ArthurYdalgo, which versions of the neighbourhood, City, and Blocks environments are you using? In the current master, the port used for RPC communication defaults to 41451. Unless you're using [...]

Not sure, and I won't be able to check it for the next couple of days

rajrana22, 2021-08-13:
Hi, I'm also having this issue and I have a feeling that it might be due to the port that I'm specifying. Where do you find out which port your specific environment uses?

ArthurYdalgo, 2021-08-24:
> Hi, I'm also having this issue and I have a feeling that it might be due to the port that I'm specifying. Where do you find out which port your specific environment uses?

I don't know. All I did was try both of them because I noticed one of the example codes that I used was using a different from the specified in the documentation.

rajrana22, 2021-08-24:
I figured out a solution. I used the flag "-RenderOffscreen" when running binaries/the UE4 Editor, and it worked perfectly.

tanyaapaul, 2022-11-27:
I am having this issue as well but with : 
WARNING:tornado.general:Connect error on fd 7: ECONNREFUSED
WARNING:tornado.general:Connect error on fd 7: ECONNREFUSED
WARNING:tornado.general:Connect error on fd 7: ECONNREFUSED
WARNING:tornado.general:Connect error on fd 7: ECONNREFUSED
WARNING:tornado.general:Connect error on fd 7: ECONNREFUSED