microsoft/AirSim #1955: build.cmd does't build anything except for a welcome info

yakunzhao, 2019-05-14:
Hello, I have been trying to build AirSim using build.cmd just as the instruction online. But when I run build.cmd through the commamd line , nothing happened except for a short welcome information:

W:\gitRepos\AirSim>build.cmd

Hello there! We just upgraded AirSim to Unreal Engine 4.18 and Visual Studio 2017.
Here are few easy steps for upgrade so everything is new and shiny:
https://github.com/Microsoft/AirSim/blob/master/docs/unreal_upgrade.md
[image]

Except for this, nothing changed. No errors, no mistakes alerts, and no profile generated (obviously). I don't know what am I missing because this is literally at the very start of the Building of AirSim. You see, 
[image]

Does anyone knows the reason of such situation or have encountered the same?

yakunzhao, 2019-05-14:
I'm using VS 2015, but I have got all the updates it needs so I suppose that shouldn't be the problem? And the Unreal version is the latest (4.18.3).

makeisbad, 2019-05-14:
之前我使用的也是vs2015，build过程中出现这个问题，就安装了vs2017，我安装的是专业版，一定要把vs2015卸载干净，用TotalUninstaller软件卸载vs2015，刚好最近在搞，环境已经搭建好了，看你昵称，姑且认为是国人。

yakunzhao, 2019-05-15:
> 之前我使用的也是vs2015，build过程中出现这个问题，就安装了vs2017，我安装的是专业版，一定要把vs2015卸载干净，用TotalUninstaller软件卸载vs2015，刚好最近在搞，环境已经搭建好了，看你昵称，姑且认为是国人。

谢谢！我会试一下17版本能否解决我的问题~

NextSim, 2019-05-31:
When you build you don't use the normal command prompt... you have to use the developer command prompt for Visual Studio.

closed 2020-08-11 by madratman