microsoft/AirSim #1956: Finding the correct position: Multicopter following a leader overtake him

Vorkosik, 2019-05-14:
## Idea:
My main goal is to try to implement basic formation flight that I can understand (learning the ropes) and my first step was to give two follower multicopters the position of a leader copter and then let them follow him.

[image]
_(Multicopter in front = leader, two in the back = follower)_

## Problem:
The leader follows a (waypoint) path but the followers always overtake him (see screenshot), then wait for the leader to 'catch up' to his path and then overtake him again.
They are allowed a faster velocity (to be able to follow closely and after course changes), but I naturally tried different scaling factors and also the same speed. They don't get the waypoints - only the leader position with this Python-code:
 
```

    def get_leader_position(self):
        # return self.client.getMultirotorState(vehicle_name=leader_name).kinematics_estimated.position
        return self.client.simGetVehiclePose(vehicle_name=leader_name).position

    def follow_leader(self):
        position = self.get_leader_position()
        self.future = self.client.moveToPositionAsync(position.x_val,
                                                      position.y_val,
                                                      position.z_val,
                                                      self.velocity*1.2,
                                                      vehicle_name=self.name)
```
As you can see in the comment above, I already tried two different methods for getting the position - which I found in the API (website and client.py). I also tried running my code for each drone in parallel but I don't really understand all the 'async'-parts and suspect that that is part of my problem(s)😉

The following code is my 'main loop' for the followers:
```
leader.follow_path()
while leader.get_position().z_val < -1:
    for drone in drone_list:
        drone.run()
    # time.sleep(0.05)
print(count)
```
And my 'run'-function:
```
    def run(self):
        if not self.started:
            self.fast_start()
            self.future.join()
        if not self.is_leader:
            self.follow_leader()
```

I'd be very thankful if you could point me in the right direction and/or help me achive some basic leader-follower formations 😄👍
Have a nice day and thank you all for AirSim!

msb336, 2019-05-14:
@Vorkosik In this setup, the follower drones will always overtake the leader, as they will be constantly following the shortest line between themselves and the goal point. 

You'll want to set up positions in 3D space relative to the leader drone as goal points for your followers, and then some form of speed controller so that they can smoothly meet their goal positions instead of moving at a constant velocity and stuttering once they reach their goal.

Vorkosik, 2019-05-14:
@msb336 Thanks for the fast reply!:smiley:
But shouldn't their goal point be the position of the leader?
If (for example) the leader wants to travel from **_X=0_** to **_X=10_** (his goal) the followers should only have his current position as goal points **(**`self.client.simGetVehiclePose(vehicle_name=leader_name).position`**)** - so if he is at, let's say **_X=5_** they shouldn't want to travel to **_X=10_** but **_X=5_**, or do they? (if they are at **_X=5_**, the leader may be at **_X=6_** and they start moving there etc.)
Does "**_simGetVehiclePose_**" maybe return some kind of goal point and not really the current position at the moment of the _getPose_-request?
I hoped that my followers would be constantly checking the 'at the moment' position of the leader and only move to this position until the check-loop runs again and not further down the path.

Maybe I'm missing your point so sorry for that!:confused: 
I knew that I'll be needing a speed controller later on (stuttering is acceptable for me at first - until I learn to program the controller, that is:grin:) and 3D-space positions around the leader, but wasn't sure how to get the exact position of the leader (as all 'copters seem to have their own coordinates - e.g. fly to the same position but are separated by the spawning distance/positions).
Thanks again for your time!:blush:

msb336, 2019-05-14:
`simGetVehiclePose` returns the current position of the drone relative to it's own origin. One of the steps in your algorithm will have to be a localization step, in which the follower drones determine their own location  with reference to the leader.

If you have the waypoint of the followers set at the leader location, they will inevitably catch up with the leader. Especially in a simulation environment. If they don't catch up with the leader, they will almost definitely collide with the other drones in sim. Unless you are building kamikazi hunter drones (which sound pretty dope), you probably want to make sure they are not all heading to the same point.

I would set up your follower control loop in a fashion like this:
1. Determine their desired waypoint relative to the leader drone
2. Determine their desired waypoint relative to themselves
3. Determine control input
4. Repeat

The possibilities for the types of formations you can build are limited only by your imagination (and by the amount of drones you can fit in the scene before performance starts to drop). Put them in a straight line, or a V formation, or a square, or a cube! Once you get all of the transformations down, it will be easy to add in new formations.

Vorkosik, 2019-05-14:
Yes, my first thought was to see how they crash in the simulator, so I purposely wanted to code kamikazi hunter drones to start the project off :grin:
But as one can see: I didn't succeed...
After all I just don't understand how my code enables the followers to overtake the leader and arrive first at arbitrary points. How is that possible if they only get the 'current position' of the leader and still have to fly there? (they don't simply overshoot - I tested that with different velocities and really slow speeds still show this behavior)
The reason that they (currently) don't crash is certainly my implementation, which doesn't account for the different origins of each drone you mentioned, but I thought that I at least could get them to arrive at the leaders old **_-and not future-_** (relative) position (with some delay of course).

If I get them to crash after implementing your mentioned follower control loop, then I'm going for a straight line formation:wink:
Currently they move like a **"L"** (movement from the right to left) when they overtake the leader and the line only exists because of the initial origin points which are in a line (the sides next to each other).

I uploaded a short video clip to hopefully get the confusing behavior (for me) across:yum:
-> https://youtu.be/8cQs9byZ7dI (right at the start is some overtaking-action and at the end [t=40s] of the video also)
Maybe it's some update loop in the background thats responsible? My loop should run continuously because it has no delays or anything: 
```
while leader.get_position().z_val < -1:
    for drone in drone_list:
        drone.run() # here the aforementioned checks/get will be requested
print(count)
```

closed 2021-08-30 by jonyMarino