microsoft/AirSim #1963: UE4 plugins not building and Blocks not building *.sln after update_from_git

yrazin, 2019-05-15:
On Windows 10, VS 2017, Unreal 4.18, trying to set AirSim up for the first time.

I have gone through the instructions for building on Windows as well as all related issues and have not been able to resolve this.

First I have built Airsim multiple times (incl. re-cloning repo, cleaning) and that seems fine, returns no errors.  When I try to add the Blocks .uproject to UE4.18 projects it says UE4Editor-AirSim.dll and UE4Editor-Blocks.dll are missing (I thought building AirSim populated the .dlls in Plugins. No?)

 When I try to update_from_git.bat (in a separate attempt, before trying to add project) I get the 
```
D:\Program Files (x86)\Microsoft Visual Studio\2017\Community\Github\AirSim\Unreal\Environments\Blocks>powershell -command "& { (Get-ItemProperty 'Registry::HKEY_CLASSES_ROOT\Unreal.ProjectFile\shell\rungenproj' -Name 'Icon' ).'Icon' } > gen_temp.tmp"
Get-ItemProperty : Cannot find path 'HKEY_CLASSES_ROOT\Unreal.ProjectFile\shell\rungenproj' because it does not exist.
At line:1 char:6
+ & { (Get-ItemProperty 'Registry::HKEY_CLASSES_ROOT\Unreal.ProjectFile ...
+      ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
    + CategoryInfo          : ObjectNotFound: (HKEY_CLASSES_RO...hell\rungenproj:String) [Get-ItemProperty], ItemNotFo
   undException
    + FullyQualifiedErrorId : PathNotFound,Microsoft.PowerShell.Commands.GetItemPropertyCommand

```
I know others have reported similar issues in the past, but I have tried restarting, uninstalling and reinstalling UE, cleaning and rebuilding AirSim and Blocks.   That registry by the way does not exist - these is not Unreal.ProjectFile key at all, forgot sub-keys.

Any ideas?

Thanks you!

msb336, 2019-05-16:
Here are my usual go-tos when unreal becomes a thorn in my side:
1. Delete all Binary and Intermediate folders in both your project folder and in `Plugins/AirSim`

2. Copy `Unreal/Plugins` folder into `Unreal/Environments/Blocks` folder manually after running `build.cmd`. If unreal complains that plugins are missing or need to be rebuilt, click ok.

To generate a visual studio solution file, right click `blocks.uproject` and click `Generate Visual Studio project files`.

msb336, 2019-06-04:
closing due to lack of activity. If you are still having problems related to this issue, please re-open the thread

closed 2019-06-04 by msb336

Nicolas-Ng, 2020-03-31:
I am having this issue almost identically. I have been able to successfully generate the Blocks.sln file, however when I open it in VS2017 15.9 and attempt to run it in Develop mode it tells me the following:

> There were build errors. Would you like to continue and run the last successful build? Y/N

Choosing yes leads to this error:

> The following modules are missing or built with a different engine version: UEEditor-Blocks.dll, UE4Editor-AirSim.dll
> Would you like to rebuild them now? Y/N

Choosing yes results in
> Blocks could not be compiled. Try rebuilding from source manually." and for all my efforts I can't figure out how to get past this.

I've done the same after manually deleting the Binary and Intermediate folders, and copying in the Plugins folder as @msb336 suggested. No luck.

Vampireqyz, 2022-01-13:
> 我几乎同样遇到了这个问题。我已经能够成功生成 Blocks.sln 文件，但是当我在 VS2017 15.9 中打开它并尝试在开发模式下运行它时，它会告诉我以下信息：
> 
> > 存在构建错误。您想继续并运行最后一次成功的构建吗？是/否
> 
> 选择 yes 会导致此错误：
> 
> > 以下模块缺失或使用不同的引擎版本构建：UEEditor-Blocks.dll、UE4Editor-AirSim.dll
> > 您现在要重新构建它们吗？是/否
> 
> 选择是导致
> 
> > 无法编译块。尝试手动从源代码重建。”尽管我付出了所有努力，但我无法弄清楚如何克服这一点。
> 
> 在手动删除 Binary 和 Intermediate 文件夹并按照@msb336 的建议复制到 Plugins 文件夹后，我也做了同样的事情。没有运气。
> 
> 我会按照建议重新打开这个线程，但鉴于这是我第一次在 github 上发帖，我不知道该怎么做。任何帮助，将不胜感激。

yeah,we have the same problem now.so do you have already done this problem,would you like to share your solution with us.

Vampireqyz, 2022-01-13:
> 我几乎同样能够遇到这个问题。我已经成功生成了 Blocks.sln 文件，但是当我在 VS2017 15.9 中打开并尝试在开发模式下运行时，它会告诉我以下信息：
> 
> > 存在构建错误。您想继续并运行最后一次成功的构建吗？是/否
> 
> 选择yes会导致此错误：
> 
> > 下面这些.dll 或使用您的不同引擎的版本编译吗？是/否
> > 现在要重新构建
> 
> 选择是导向
> 
> > “虽然我无法编译所有的努力，但我无法获得成功。”
> 
> 在手动删除 Binary 和 Intermediate 文件夹并没有按照@msb336 的建议复制到插件文件夹后，我做了同样的事情。运气也。
> 
> 我会根据我的建议打开这个话题，但这是我在github上建议的一次，我不知道该怎么做。

looking forward to your reply early.

Chen1DONG, 2024-04-24:
我也遇到了相同的问题。QAQ