microsoft/AirSim #1964: Unity part of the Build FAILED. error  C2259

Vahe987, 2019-05-15:
# Unity part of the Build FAILED. error  C2259

## Build of the project succeeded,but after that when I run the 'build.cmd' in Unity folder it fails.
in the simmodebase.cpp line 18
world_sim_api_.reset(new WorldSimApi(this, vehicle_name_));
I was able to build if I ignore this part, like doing "world_sim_api_.reset()"instead,
but this should be fixed, don't know what it can break now
## here is the error message
Build FAILED.

"c:\Mila\AirSim\Unity\AirLibWrapper\AirsimWrapper.sln" (Clean;Build target) (1) ->
"c:\Mila\AirSim\Unity\AirLibWrapper\AirsimWrapper\AirsimWrapper.vcxproj.metaproj" (default target) (2:2) ->
"c:\Mila\AirSim\Unity\AirLibWrapper\AirsimWrapper\AirsimWrapper.vcxproj" (default target) (5:2) ->
(ClCompile target) ->
  c:\mila\airsim\unity\airlibwrapper\airsimwrapper\source\simmode\simmodebase.cpp(18): error C2259: 'WorldSimApi': cannot instantiate abstract class [c:\Mila\AirSim\Unity\AirLibWrapper\AirsimWrapper\AirsimWrapper.vcxproj]

    0 Warning(s)
    1 Error(s)

galgyeony, 2019-05-16:
I encontered the same problem and still do not know how to solve it.
Hope to see a proper solution soon.

msb336, 2019-05-16:
@galgyeony looks like [this commit](https://github.com/microsoft/AirSim/pull/1925/files#diff-5eb19662930009010d368bddc989b729R45) added an API to WorldSimApi that is only defined in the Unreal Plugin. This means there is a pure virtual function in WorldSimApiBase that is never defined in the unity plugin.

I'll submit a fix today, but in the meantime, rolling back to the previous commit should unbreak your build.

Vahe987, 2019-05-16:
@msb336 , thank you for taking a look at this short notice, will wait for the fix!

msb336, 2019-05-16:
closing this issue, as a fix has been commited. please re-open if you are continuing to have problems

closed 2019-05-16 by msb336

Vahe987, 2019-05-16:
I've tried, it works, thank you @msb336

galgyeony, 2019-05-17:
It works! Thank you @msb336 for your quick support.

adamdarko, 2019-08-29:
I am having this issue and it is still causing the drone and the car to not be controlled.
I have installed unity from the visual studio installer and not the unity hub, though im not sure if that's the issue.

dion26, 2020-06-09:
How to get this fix? I am trying to install the 1.2.2 Version and get this exact error.