microsoft/AirSim #1965: Question regarding launching ArduCopterSolo Vehicle

rajat2004, 2019-05-16:
Hi,
I'm working on creating the SITL backend for AirSim in ArduPilot and as the first step for that, I've written an initial connector for AirSim which supports the Solo connector added in #1387.
It can be found in my Ardupilot branch if anyone's interested - https://github.com/rajat2004/ardupilot/tree/sitl-airsim

My question is regarding the `settings.json` file for launching the Solo. My current settings file is as follows - 

```
{
  "SettingsVersion": 1.2,
  "SimMode": "Multirotor",
  "Vehicles": {
      "Solo": {
	  "VehicleType": "arducoptersolo",
	  "UseSerial": false
      }
  }
}
```

After starting the Airsim plugin in Unreal engine, I'm getting the following message
[image]

Most probably the problem is with the connector which I've written but just wanted to confirm that the settings.json I'm using is correct or if any other fields are also required.

The sockets from the Ardupilot side are opening properly and ports, ip address are all set to the default values in [AirSimSettings.hpp](https://github.com/microsoft/AirSim/blob/master/AirLib/include/common/AirSimSettings.hpp#L248)

System Details - Ubuntu 16.04, Unreal Engine 4.18, AirSim Latest master

rajat2004, 2019-05-17:
@KerryMoffittRtn Could you help me out here?

Ardupilot is binding the sockets correctly and Airsim is also connecting to the sockets (after commenting out the GPS log message which was getting printed continuously, shows that Airsim is trying to send the sensor packet)
But the Ardupilot SITL isn't receiving any data

I haven't written a small piece of code till now for converting the data from little-endian to the host endianess in the Ardupilot backend but I don't think that's causing the problem.
Anyways, I'll try and write that tonight to make sure of this.

A small piece of the log messages generated by Airsim, I don't think the warnings are relevant, but just posting them here

```
[2019.05.17-15.03.57:725][203]LogActor: Warning: BP_CameraDirector_C /Game/FlyingCPP/Maps/UEDPIE_0_FlyingExampleMap.FlyingExampleMap:PersistentLevel.CameraDirector has natively added scene component(s), but none of them were set as the actor's RootComponent - picking one arbitrarily
[2019.05.17-15.03.57:894][203]LogTemp: Using UDP port 14560, local IP 127.0.0.1, remote IP 127.0.0.1 for sending sensor data
Using UDP port 14560, local IP 127.0.0.1, remote IP 127.0.0.1 for sending sensor data
[2019.05.17-15.03.57:894][203]LogTemp: Using UDP port 14556 for receiving rotor power
Using UDP port 14556 for receiving rotor power
[2019.05.17-15.03.58:301][203]LogScriptCore: Warning: Script Msg: Attempted to access index 0 from array 'CallFunc_GetAllActorsOfClass_OutActors' of length 0 in '/AirSim/Weather/WeatherFX/WeatherActor.WeatherActor_C:ExecuteUbergraph_WeatherActor'!
[2019.05.17-15.03.58:584][203]LogTemp: Solo
```

My current `settings.json` (added the IP and port fields just to make sure) - 
```
{
	"SettingsVersion": 1.2,
	"SimMode": "Multirotor",
	"Vehicles": {
		"Solo": {
			"VehicleType": "arducoptersolo",
			"UseSerial": false,
			"AllowAPIAlways": false,
			"UdpIp": "127.0.0.1",
			"UdpPort": 14560,
			"SitlIp": "127.0.0.1",
			"SitlPort": 14556 
		}
	}
}
```

rajat2004, 2019-07-19:
Closing this issue.
I have written an interface in Ardupilot compatible with this Solo API present in AirSim, my branch is here-
https://github.com/rajat2004/ardupilot/tree/airsim-soloapi
This also has a Readme for using the interface.
It won't be updated any further and is just kept if anyone wants to try it out.

A different API based on JSON packets and implementing Lock-Step Scheduling is now in ArduPilot master (PR https://github.com/ArduPilot/ardupilot/pull/11367) with follow-up PR https://github.com/ArduPilot/ardupilot/pull/11835 for Lidar Support.

Corresponding PR in AirSim - https://github.com/microsoft/AirSim/pull/2075

A PR for adding the Wiki page in Ardupilot Dev documentation is here - https://github.com/ArduPilot/ardupilot_wiki/pull/1846

closed 2019-07-19 by rajat2004

zimmy87, 2021-03-11:
Hey @rajat2004, after you contributed the ArduPilot support, is there any advantage to keeping the Solo API? We're considering removing it to clean up the codebase and make it more understandable for newcomers.

rajat2004, 2021-03-12:
@zimmy87 IMO, yeah, the old Solo API can be removed. The ArduPilot interface which I initially wrote for that is very old, and might not even compile. I can't think of any reason to use the Solo vehicle instead of the AP Copter, Rover vehicles