microsoft/AirSim #1967: multi_agent_drone.py Not Working: Vehicle API for 'Drone1' is not available

PatrickUtz, 2019-05-16:
Hello,

I am unable to successfully run multi_agent_drone.py since I keep getting the following error:

`msgpackrpc.error.RPCError: rpclib: function 'enableApiControl' (called with 2 arg(s)) threw an exception. The exception contained this information: Vehicle API for 'Drone1' is not available. This could either because this is simulation-only API or this vehicle does not exist.`

I read all the similar issues that people posted yet am unable to fix it. I am running the latest version of AirSim and made sure to make appropriate changes to settings.json. I have attached screenshots of the full traceback along with my settings.json file. Thank you. 

OS: Ubuntu 18.04.2 LTS
Python: 3.7.3

[image]
[image]

PatrickUtz, 2019-05-21:
*SOLVED*
Turns out that I was editing the settings.json file under /AirSim/docker location. This IS NOT the right settings.json file and instead you are supposed to edit the settings.json file located under ~/Documents/AirSim (for Linux). Made the appropriate changes and now I see two drones in AirSim!

closed 2019-05-21 by PatrickUtz

sandilyasg, 2023-10-10:
@PatrickUtz Can you please provide detail on how you launched this? I am using the UnityDemo project and I am following the same steps as you are but I launch the SimModeSelector scene. I tried with MultiDroneDemo but it still does not work.

bututoubaobei-0108, 2024-03-19:
Remove the argument concerning the drone_name works for me. It is a way used by DRLwithTL project presented in https://towardsdatascience.com/deep-reinforcement-learning-for-drones-in-3d-realistic-environments-36821b6ee077