microsoft/AirSim #1975: IMU Angles (new API) vs Camera Angles  + Gimbal BUG

stavBodik, 2019-05-21:
My goal is to get the angles of the camera using IMU with gimbal , I know 2 methods which should give me this info :

float yawCamera, pitchCamera, rollCamera;
VectorMath::toEulerianAngle(image_info.camera_orientation, pitchCamera, rollCamera, yawCamera);

or

float yawCamera1, pitchCamera1, rollCamera1;
msr::airlib::ImuBase::Output imu_data = client.getImuData();
VectorMath::toEulerianAngle(imu_data.orientation, rollCamera1, pitchCamera1, yawCamera1);

The first thing missing is that IMU API (second method) gives the orientation of the body and not of the camera,
**In image processing we care about the orientation of the camera not the body.**
(the same for accelerations)

The second thing is probably a bug , when using GIMBAL with 100% fixed angles , 
This methods should return the orientation of the camera with respect to the GIMBAL .

"Gimbal": {
      "Stabilization": 1,
      "Pitch": -90, "Roll": 0, "Yaw": 0
	}

This methods should return ~ 0,-90,0 constantly.

stavBodik, 2019-06-05:
No one want to fix it ?

ACLeighner, 2020-06-18:
Ill add that the problem seems to be that simGetCameraInfo returns the camera pose combined with the body pose,  so you are getting the camera pose in world coordinates rather than relative to the body.  I haven't tried it yet but you might be able to simply subtract the body pose from what simGetCameraInfo gives you.  Its on my list of things to try so ill come back if I figure it out.