microsoft/AirSim #1979: [Feature Request] Record LiDar data in manual control

MohamedAliRashad, 2019-05-23:
I was wandering if there was a way to record my driving with the LiDar data gathered in real time without the need of a python script ... like what's happening with the camera data.

msb336, 2019-05-23:
There is no in-app gui to save LiDAR data, but you can record lidar data while in manual control. By running a python script like this:
```
client = airsim.CarClient()
while True:
    // collect lidar data
    timestamp = time.time()
    lidar = client.getLidarData()
    // save to file
```

You can collect lidar data while using keyboard controls for your vehicle. Unfortunately, a Python or c++ script is required to do this.

MohamedAliRashad, 2019-05-24:
I hoped a feature like this could be implemented and controlled through ```settings.json```.
we can even implement a small GUI for users in the start of AirSim asking them what sensors they want to use in data recording and based on the choices a ```settings.json``` file would be generated.

closed 2019-06-04 by msb336