microsoft/AirSim #1980: Airsim Environments in Unreal Engine Marketplace

ganscool86, 2019-05-23:
Hi, Is it possible to get the Airsim Environments like Africa, plains and others through the Unreal Engine Marketplace for a price. Are you planning to sell it in the future. Currently the precompiled environments binaries are outdated with the latest airsim codes and if we can get the environment alone then it will help us build/ test our own project more efficiently. Thanks.

MahdiulChowdhury, 2020-01-14:
I was wondering the same, I liked the environment and if I really would like to open it in Unreal editor, create new plug in for sim and build it.

Is there any update on this ?

jonyMarino, 2020-08-14:
As far as I know this is out of Airsim's plans so far. I will share it anyway