microsoft/AirSim #1983: How to uninstall airsim cleanly on windows?

wtq1996, 2019-05-24:
Dear all,

I think I am facing version problem now, so I want to uninstall the airsim which was installed by running 'build.cmd' in Visual Studio Command Window.

So, here is the whole story. I downloaded the pre-compiled environment binaries (CityEnviron, etc) and all I need to do is just 'pip install airsim' in my conda environment (which is v1.2.4), then everything works perfectly except I found that it's not capable to placing car and multirotor in the scene at the same time. So, I downloaded this customed airsim repo
[https://github.com/kshitiz38/AirSim](url), in order to run the build.cmd file, I installed Visual Studio. Then, I met a same problem which occured previously on another computer where I followed the official installing doc for linux system (UE4, clone airsim repo, run build.sh). The problem is "msgpackrpc.error.RPCError: rpclib: server could not find function 'getImuData' with argument count 2."(or getGps, etc). I guess  this is a version problem, cause the v1.2.4 pip installed airsim worked perfectly before. So I tried to uninstall the airsim installed by 'build.cmd' cleanly, but I just didn't figure out, I didn't see any uninstall file in the airsim folder, so I just simply deleted the whole folder, but seems like some installed file is still in my computer, so I couldn't get back to the previous perfectly running status.

So, my question is that how to uninstall airsim cleanly if it was installed by running 'build.cmd' .

Thanks alot

msb336, 2019-05-24:
@wtq1996
getImuData is a very new API, most of our environment Binaries are not caught up to that release yet, but don't worry, we're working on that.

Deleting your repo, or running `git clean -fdx` should both be sufficient at uninstalling the repoed version of airsim. However, to uninstall the Python client, you will also have to run the following command:

`pip uninstall airsim`

wtq1996, 2019-05-28:
> @wtq1996
> getImuData is a very new API, most of our environment Binaries are not caught up to that release yet, but don't worry, we're working on that.
>
> Deleting your repo, or running `git clean [...]

Hi, thanks for your reply! But the solution seems doesn't work for me. Since after I uninstalled airsim in my conda environment ('pip uninstall airsim') and deleted all the Airsim-master folder except the PythonClient subfolder, if I just simply 'import airsim', it will show no module named airsim, but if I use 'import setup_path; import airsim', then everything works same as before (all basic api works well, but api like getIMU,getDistance, setWeather, setTimeofday don't work). If I install airsim using pip install airsim again, and simply use import airsim, same error excepy the client.py's path has been changed. I guess maybe pip install airsim only download several interface files like the files in PythonClient/airsim, maybe the problem is caused by the msgpackrpc module cause the error is related to that. By the way, my another computer using ubuntu system followed the official build on linux tutorial, basic and setWeather, setTimeofday works fine, but the getIMU, getGPs, getDistance,etc do not work, I don't know what caused this difference. I really need getDistance() cause I want to get the distance from the closet obstacle as my reinforcement learning reward(I guess that's what getDistance() returns,right?)

Case1:
import setup_path 
import airsim
# connect to the AirSim simulator
client = airsim.CarClient()
client.confirmConnection()
client.enableApiControl(True)
car_controls = airsim.CarControls()
dis = client.getDistanceSensorData()
s = pprint.pformat(dis)
print("dis_data: %s" % s)

File "car_time_of_day.py", line 81, in <module>
    timeOfDayTest.execute()
  File "car_time_of_day.py", line 36, in execute
    dis = self.client.getDistanceSensorData()
  File "/home/control02/anaconda3/envs/pytorch/lib/python3.6/site-packages/airsim/client.py", line 157, in getDistanceSensorData
    return DistanceSensorData.from_msgpack(self.client.call('getDistanceSensorData', distance_sensor_name, vehicle_name))
NameError: name 'DistanceSensorData' is not defined

Case2:
import airsim
# connect to the AirSim simulator
client = airsim.CarClient()
client.confirmConnection()
client.enableApiControl(True)
car_controls = airsim.CarControls()
dis = client.getDistanceSensorData()
s = pprint.pformat(dis)
print("dis_data: %s" % s)

File "/home/control02/Downloads/AirSim-master/PythonClient/airsim/client.py", line 160, in getDistanceSensorData
    return DistanceSensorData.from_msgpack(self.client.call('getDistanceSensorData', distance_sensor_name, vehicle_name))
NameError: name 'DistanceSensorData' is not defined

wtq1996, 2019-05-28:
> @wtq1996
> getImuData is a very new API, most of our environment Binaries are not caught up to that release yet, but don't worry, we're working on that.
>
> Deleting your repo, or running `git clean [...]

Sorry for my confusing explainlation, but I seems found the reason. It has nothing to do with my airsim, it's just the binaries that I downloaded for windows and linux may not support the same api even for the same environment like Neighborhood cause they have different release date. 
But unfortunately, I didn't find any binary environment that can support all the APIs that I mentioned before.
I'll summarize my downloaded version for those binaries and the corresponding problems:

v1.2.2 windows 2019/5/15(only city provided):
City: getDistanceSensorData() cannot use,  Error: DistanceSensorData not defined
        other APIs work perfectly fine

v1.2.1windows 2019/1/24: 
Neighborhood, Coastline: getImudata, getGps, getDistance, etc cannot use.
                                          setWeather, setTimeofDay cannot use.
Blocks: same as before except setTimeofDay works

v1.2.0 linux 2019/1/12:
City:
I think this zip.001 and zip.002 files are destroyed, cause I cannot extract this while V1.2.2 windows City can be extracted normally.
Neighborhood:
Everything works fine, except getDistanceSensorData() cannot use,  Error: DistanceSensorData not defined.

I am trying to train the car to drive autonomously on the road while avoiding collision in whether City or Neighborhood, so I need to get the distance from the closet obstacle as each state's reward. Can you tell me how can I get that? Also, apart from the distance from the closet obstacle, is there any possibility to get the distance from the center-line or side-line of the road? Thakns a lot!

jonyMarino, 2020-08-17:
@tianqi-wang1996 can you try your code with latest version please?