microsoft/AirSim #1986: Collision Avoidance : has_collided returns False even on colliding with a certain object

mish1102, 2019-05-27:
I am using AirSim NH environment to train my drone using RL with CNTK. While the drone hits the certain object in the scene, still in background has_collided parameter is assigned False. 
Kindly look below the Collision_Info :

('Collision_data', <instance> {   u'has_collided': False,
    u'impact_point': <instance> {   u'x_val': 0.0,
    u'y_val': 0.0,
    u'z_val': -0.14649589359760284},
    u'normal': <instance> {   u'x_val': 1.6051125272670275e-13,
    u'y_val': -1.1454494019957906e-09,
    u'z_val': -1.0},
    u'object_id': 171,
    u'object_name': u'Road_89',
    u'penetration_depth': 0.011044786311686039,
    u'position': <instance> {   u'x_val': 0.0,
    u'y_val': 0.0,
    u'z_val': -0.14649589359760284},
    u'time_stamp': 1558946282693258752})

Help is much appreciated.

msb336, 2019-06-04:
@mish1102 Can you provide a screenshot of the occurrence? A definition of the object in question will allow a quicker debugging process.

Hoshea7, 2019-08-01:
Hi @msb336, 

I meet the same problem when I am using multirotor. 
Here is the screenshot that the problem I meet. I set the Airsim Environment in 5X clock speed when I take action to move forward, and the simulator returns the information that meets the collision but the function "collision_info.has_collided" return False. 
[image]
I wonder if the problem is because the response time to detect the collision information by python client to be delayed? By the way, or could I get the information about the collision count number? so that I also can know if the vehicle was been through collision before or not?

jonyMarino, 2020-09-10:
may be related to #1004?

alirezadizaji, 2020-10-31:
I have same issue

rolanvc-upd, 2021-08-18:
I have the same issue

riyadfebrian, 2021-12-21:
yeet, have the same problem, i think it was the `has_collided` bug when the car or quadrotor hit some object,  it's not changing the flag into `True`. so that was the problem.

what my trick to solve this issue is to add more parameter to make sure the car was in collision. 
take a look at collision data:

`('Collision_data', { u'has_collided': False,  

u'impact_point': { u'x_val': 0.0,  
u'y_val': 0.0,
u'z_val': -0.14649589359760284},
u'normal': { u'x_val': 1.6051125272670275e-13,
u'y_val': -1.1454494019957906e-09,
u'z_val': -1.0},
u'object_id': 171,
u'object_name': u'Road_89',
u'penetration_depth': 0.011044786311686039,
u'position': { u'x_val': 0.0,
u'y_val': 0.0,
u'z_val': -0.14649589359760284},
u'time_stamp': 1558946282693258752})`

use `normal.x_val` as your second parameter. 
in example : 
`collision = self.drone.simGetCollisionInfo()`
`self.state["collision"] = collision.has_collided`
`self.state["object_normal_position"] = collision.normal.x_val`

checking 
`if self.state['collision'] or self.state['object_normal_position'] != 0:
            reward = -10`