microsoft/AirSim #1988: vehicle %s was not initialized

PeixinLiu, 2019-05-28:
Hello everyone:
 I'm trying to use HIL and a Joystick controller .but it shows that "vehicle %s was not initialized" ,as follows:
[image]

And this is my settings:
    {
      "WindowID": 2,
      "ImageType": 6,
      "CameraName": "4",
      "Visible": true
    }
  ],
  "SimMode": "Multirotor",
  "DefaultVehicleConfig": "PX4",
  "Vehicles": {
    "PX4": {
      "VehicleType": "PX4Multirotor",
      "LogViewerHostIp": "127.0.0.1",
      "LogViewerPort": 14388,
      "OffboardCompID": 1,
      "OffboardSysID": 134,
      "QgcHostIp": "127.0.0.1",
      "QgcPort": 14550,
      "SerialBaudRate": 115200,
      "SerialPort": "*",
      "SimCompID": 42,
      "SimSysID": 142,
      "SitlIp": "127.0.0.1",
      "SitlPort": 14556,
      "UdpIp": "127.0.0.1",
      "UdpPort": 14560,
      "UseSerial": true,
      "VehicleCompID": 1,
      "VehicleSysID": 135,
      "Model": "Generic",
      "LocalHostIp": "127.0.0.1"
    }
  }
}

The most gratitude

xiaolin360, 2019-05-30:
I encounter the same problem as you when i plug in the Pixhawk 4 hardware, but this problem disappears when i plug in the Pixhawk 2.4.8 hardware.
And this is my settings:
{
  "SettingsVersion": 1.2,
  "SimMode": "Multirotor",
  "Vehicles": {
    "PX4": {
      "VehicleType": "PX4Multirotor"
    }
  }
}
I don't know how to solve this problem and am seeking for more help.

heleidsn, 2019-06-08:
Hi, you need to close your Qgroundcontrol after the initial setup. Then you can try to connect the Pixhawk and restart your UE project.

Hope it can help you.

kaioforte, 2019-10-03:
There is another solution? Even when I close Qgroundcontrol after the initial setup, the problem persists "vehicle %s was not initialized". This happens on Windows, if I test on Linux it says "connected to px4" but Unreal engine freezes and crashes. I use a pixhawk 4 mini with a firmware version of 1.8.2. I've tried with the 1.9.2 (latest stable version) but it is the same.
I saw this issue https://github.com/microsoft/AirSim/issues/1841 where @lovettchris fixes a missing module in version 1.8.2. It works, but that build is for fmu-v2 and I think it is not supported by pixhawk 4 (fmu-v5). Do you think it is a similar problem? And you could take a look in it or tell me how to do that?
Ps: I use the same settings as xiaolin360.

cnun02, 2019-10-14:
Airsim with PX4 #2246
I had the same problem. But now it is neatly solved.

Rufio-Boj, 2019-10-22:
I found a way of making it work with PX4 v5, see #2246

PeixinLiu, 2019-10-22:
> Hi, you need to close your Qgroundcontrol after the initial setup. Then you can try to connect the Pixhawk and restart your UE project.
> 
> Hope it can help you.

thanks for your help,I will try it

PeixinLiu, 2019-10-22:
> Airsim with PX4 #2246
> I had the same problem. But now it is neatly solved.

ok thanks ,I will try again

PeixinLiu, 2019-10-22:
> I found a way of making it work with PX4 v5, see #2246

ok thanks ,I will try again

closed 2021-06-03 by jonyMarino