microsoft/AirSim #1992: Output RGB Value Does Not Match Value Set By simSetSegmentationObjectID()

PatrickUtz, 2019-05-30:
I followed the instructions [here](https://microsoft.github.io/AirSim/docs/image_apis/#what-does-pixel-values-mean-in-different-image-types) for using `simSetSegmentationObjectID()` to set the id of my "WhiteSphere" mesh to 210. According to the docs, this should set the color of "WhiteSphere" to RGB (210, 210, 210) when taking an image using the Infrared feature with `simGetImages([airsim.ImageRequest("0", airsim.ImageType.Infrared)])`. However, as shown in the attached images, the sphere in the resulting .png file has a color of hex E9E9E9 or RGB (233, 233, 233). I figured that there is some offset between the desired value from `simSetSegmentationObjectID()` and the output .png file. Yet after setting the id of another object, I realized that the offset with its output color was different from that of the WhiteSphere and its output color. 
[image]
[image]
[image]

msb336, 2019-06-04:
@ironclownfish

Any ideas?

ironclownfish, 2019-06-04:
@PatrickUtz This looks like the same problem that was recently fixed in https://github.com/microsoft/AirSim/pull/1881.

Can you make sure your airsim install is fully updated?

PatrickUtz, 2019-06-05:
@ironclownfish I have verified that I am using the latest version of AirSim and have not been able to find out what the issue is. In about a week I will be porting everything over to my Windows machine and will check to see if I get the same issue. If not, I plan on checking/mapping each id to it's actual output grayscale value (so essentially make my own version of [this](https://microsoft.github.io/AirSim/docs/seg_rgbs.txt)).

PatrickUtz, 2019-06-05:
@ironclownfish I have the latest version installed (installed AirSim in mid-May) and continue to have the same issue. In about a week I plan on porting everything over to Windows and checking to see if the problem persists. If it does, I plan on mapping each ID to the actual value it corresponds to and make something similar to [this](https://microsoft.github.io/AirSim/docs/seg_rgbs.txt).

PatrickUtz, 2019-07-12:
@msb336 and @ironclownfish I figured out the problem. The reason that the pixel value I set using the `simSetSegmentationObjectID()` function for infrared images did not correspond to the actual grey scale value of the output image is because my `settings.json` file (found under `/home/<username>/AirSim` in linux) did not have the "CameraDefaults" setting and as such the pixel value that was set using the `simSetSegmentationObjectID()` function corresponded to a "random" grey scale value (not actually random...I went through the process of mapping out these "initialized" grey scale pixel values in the attached .txt file for reference). Now that I included the `"CameraDefaults"` and `"CaptureSettings"` in the `settings.json` file (as shown in the screenshot below), all the pixel values I pass through the `simSetSegmentationObjectID()` function now correspond to a respective greyscale value (so simSetSegmentationObjectID(Cube1, 2) causes the Cube1 mesh in the output infrared image to have a grey scale value of 2, 2, 2).
[segmentation_pixel_values.txt](https://github.com/microsoft/AirSim/files/3388116/segmentation_pixel_values.txt)
[image]

closed 2019-07-12 by PatrickUtz

espenthaem, 2019-07-16:
I'm having the same problem when using the Segmentation camera. In the Blocks environment, setting the SegmentationObjectID to 1 results in an RBG value of [199, 174, 41], instead of [153, 108, 6], according to [](https://github.com/microsoft/AirSim/blob/master/docs/seg_rgbs.txt)

My settings.json looks like: 
`{
  "SeeDocsAt": "https://github.com/Microsoft/AirSim/blob/master/docs/settings.md",
  "SettingsVersion": 1.2,
  "SimMode": "ComputerVision",
  "CameraDefaults": {
    "CaptureSettings": [
      {
        "ImageType": 0,
        "Width": 256,
        "Height": 144
      },
      {
        "ImageType": 5,
        "Width": 256,
        "Height": 144
      }
    ]
  }
}
{
  "SeeDocsAt": "https://github.com/Microsoft/AirSim/blob/master/docs/settings.md",
  "SettingsVersion": 1.2,
  "SimMode": "ComputerVision",
  "CameraDefaults": {
    "CaptureSettings": [
      {
        "ImageType": 0,
        "Width": 256,
        "Height": 144
      },
      {
        "ImageType": 5,
        "Width": 256,
        "Height": 144
      }
    ]
  }
}'

PatrickUtz, 2019-07-16:
Try adding/playing with the `"SegmentationSettings"` in the settings.json. More infor from the API docs here: https://github.com/microsoft/AirSim/blob/master/docs/settings.md
[image]

espenthaem, 2019-07-16:
Thank you for your reply. Unfortunately, I'm only able to influence the ObjectID that are used to generate the mapping. However, the RGB values that I get in my SegmentationImage do not correspond to the values listed in the [docs](https://github.com/microsoft/AirSim/blob/master/docs/seg_rgbs.txt).

PatrickUtz, 2019-07-16:
Did you add this to your `settings.json` and made sure to restart Unreal Editor?
[image]

espenthaem, 2019-07-16:
Yes. 

I use `client.simSetSegmentationObjectID("OrangeBall", 1)` in the Blocks environment, and I'm viewing the Segmentation Image using the sub window.

PatrickUtz, 2019-07-16:
Ok, I would say try using the `Infrared` feature instead and see if you get the same issue.

espenthaem, 2019-07-17:
Unfortunately I have the same issue when using the `Infrared` feature, I even get the same mapping of Segmentation ID's to grey scale values. 

I'm using the following settings:

```
{
  "SeeDocsAt": "https://github.com/Microsoft/AirSim/blob/master/docs/settings.md",
  "SettingsVersion": 1.2,
  "SimMode": "ComputerVision",
  "CameraDefaults": {
    "CaptureSettings": [
      {
        "ImageType": 0,
        "Width": 256,
        "Height": 144
      },
      {
        "ImageType": 5,
        "Width": 256,
        "Height": 144
      },
      {
        "ImageType": 7,
        "Width": 256,
        "Height": 144
      }
    ]
  },
  "SegmentationSettings": {
    "InitMethod": "",
    "MeshNamingMethod": "",
    "OverrideExisting": false
  }
}
```

PatrickUtz, 2019-07-20:
@espenthaem try putting "None" for the "InitMethod" section and then restart Unreal Editor. If this does not work, then I assume you are having the same issue I had and as such the infrared values are corresponding to another set of values which I mapped out here: [segmentation_pixel_values.txt](https://github.com/microsoft/AirSim/files/3388116/segmentation_pixel_values.txt). The left values are the ones you set in the API and the right values are the actual infrared values that are outputted in the segmented image. Hopefully this helps!