microsoft/AirSim #1997: Return full LiDAR Scan

JeremyBYU, 2019-06-03:
Currently when I call getLidarData() (cpp client) I do not always get a full lidar scan that is configured in my `settings.json`.  It seems that if I call this function at a high frequency I can get a a result of size 0. I can handle that by checking the size and timestamp.  

```cpp
auto response = client->getLidarData(lidar_name, drone_name);
if (response.point_cloud.size() > 1 && response.time_stamp != prev_timestamp) {
    prev_timestamp = response.time_stamp;
    WriteLidarData(response);
}
```

However sometimes I will get _incomplete_ scans as well.  Its almost as if the the airsim server 
(running in UE4) is simply returning whatever the current point cloud size that has currently been scanned. Is there some configuration that should be changed to allow for a providing a full point cloud scan?

Looking at the source code I'm not quite sure how this is possible because it should only update the `output` variable (which the AirSim server code pulls from when getLidarData is called) _after_ it has completed its scan. Relevant sections:

Get Lidar Data - 
https://github.com/microsoft/AirSim/blob/df70bf12bb400b0c0243b4c496f3dd5ce0b348e0/AirLib/src/api/RpcLibServerBase.cpp#L153
https://github.com/microsoft/AirSim/blob/67f438629c4ca4bcf25436d5e21e9839487a6491/AirLib/include/api/VehicleApiBase.hpp#L111
Sensor Output - https://github.com/microsoft/AirSim/blob/67f438629c4ca4bcf25436d5e21e9839487a6491/AirLib/include/sensors/lidar/LidarBase.hpp#L51
Senor Output Update - 
https://github.com/microsoft/AirSim/blob/67f438629c4ca4bcf25436d5e21e9839487a6491/AirLib/include/sensors/lidar/LidarSimple.hpp#L104

Thanks for any help!

msb336, 2019-06-04:
How many points per frame are you collecting with your LiDAR device? UnrealLidarSensor caps max points in a single to 100,000, so if your LiDAR is attempting to scan at a higher density than that you will see impartial pointclouds.

relevant code here:
https://github.com/microsoft/AirSim/blob/b0e145a20c68abc556ba3f034674961a75ef9986/Unreal/Plugins/AirSim/Source/UnrealSensors/UnrealLidarSensor.cpp#L54-L60

JeremyBYU, 2019-06-04:
Thanks for catching that, but I dont think that it is the issue: Here is a reproducible example:

Lidar Setttings:
```
        "0": {
          "SensorType": 6,
          "Enabled": true,
          "NumberOfChannels": 16,
          "RotationsPerSecond": 5,
          "PointsPerSecond": 28800,
          "X": 0.0,
          "Y": 0,
          "Z": 0,
          "Roll": 0,
          "Pitch": -30.0,
          "Yaw": 0,
          "VerticalFOVUpper": -0,
          "VerticalFOVLower": -30,
          "HorizontalFOVStart": -50,
          "HorizontalFOVEnd": 50,
          "DrawDebugPoints": true,
          "DataFrame": "SensorLocalFrame"
        }
```

This should provide 1600 points in one full scan (28800 * (100 deg/360deg) / 5) . Note that a full scan is only between (-50 - +50) degrees).  

Source Code:
```cpp
#include "common/common_utils/StrictMode.hpp"
STRICT_MODE_OFF
#ifndef RPCLIB_MSGPACK
#define RPCLIB_MSGPACK clmdep_msgpack
#endif // !RPCLIB_MSGPACK
#include "rpc/rpc_error.h"
STRICT_MODE_ON

#include "vehicles/multirotor/api/MultirotorRpcLibClient.hpp"
#include "common/common_utils/FileSystem.hpp"
#include <iostream>
#include <chrono>

int main() 
{
    using namespace msr::airlib;

    msr::airlib::MultirotorRpcLibClient client;
    
    try {
        client.confirmConnection();

        std::cout << "Press Enter to arm the drone" << std::endl; std::cin.get();
        client.enableApiControl(true);
        client.armDisarm(true);

        std::cout << "Press Enter to takeoff" << std::endl; std::cin.get();
        float takeoffTimeout = 5; 
        client.takeoffAsync(takeoffTimeout)->waitOnLastTask();

        // switch to explicit hover mode so that this is the fall back when 
        // move* commands are finished.
        std::this_thread::sleep_for(std::chrono::duration<double>(5));
        client.hoverAsync()->waitOnLastTask();

        std::cout << "Counting Lidar Return Size" << std::endl; std::cin.get();
        // moveByVelocityZ is an offboard operation, so we need to set offboard mode.
        TTimePoint prev_timestamp = 0;
        for(int i=0; i<20;)
        {
            auto response = client.getLidarData("0", "");
            if (response.point_cloud.size() > 2 && response.time_stamp != prev_timestamp) {
                prev_timestamp = response.time_stamp;
                std::cout << "Time Stamp: " << prev_timestamp << "; PC size: " << response.point_cloud.size() / 3 << std::endl;
                i++; //increment
            }
            std::this_thread::sleep_for(std::chrono::milliseconds(10));
        }

        client.hoverAsync()->waitOnLastTask();

        std::cout << "Press Enter to land" << std::endl; std::cin.get();
        client.landAsync()->waitOnLastTask();

        std::cout << "Press Enter to disarm" << std::endl; std::cin.get();
        client.armDisarm(false);

    }
    catch (rpc::rpc_error&  e) {
        std::string msg = e.get_error().as<std::string>();
        std::cout << "Exception raised by the API, something went wrong." << std::endl << msg << std::endl;
    }

    return 0;
}
```

Output:
```
Waiting for connection - X
Connected!

Client Ver:1 (Min Req:1), Server Ver:1 (Min Req:1)
Press Enter to arm the drone

Press Enter to takeoff

Counting Lidar Return Size

Time Stamp: 1559671588298269184; PC size: 1216
Time Stamp: 1559671588400271360; PC size: 384
Time Stamp: 1559671588502273536; PC size: 1328
Time Stamp: 1559671588604275712; PC size: 272
Time Stamp: 1559671588706277888; PC size: 1456
Time Stamp: 1559671588808280064; PC size: 160
Time Stamp: 1559671588910282240; PC size: 1568
Time Stamp: 1559671589012284416; PC size: 32
Time Stamp: 1559671589114286592; PC size: 1600
Time Stamp: 1559671589318290944; PC size: 1600
Time Stamp: 1559671589522295296; PC size: 1600
Time Stamp: 1559671589726299648; PC size: 1616
Time Stamp: 1559671589930304000; PC size: 1600
Time Stamp: 1559671590134308352; PC size: 1600
Time Stamp: 1559671590338312704; PC size: 1600
Time Stamp: 1559671590542317056; PC size: 1616
Time Stamp: 1559671590746321408; PC size: 1600
Time Stamp: 1559671590950325760; PC size: 1600
Time Stamp: 1559671591154330112; PC size: 1600
Time Stamp: 1559671591358334464; PC size: 1600

```

I think I must be missing something but I'm just not sure what it is. Thanks!

ironclownfish, 2019-06-06:
I'm currently looking into this.

ironclownfish, 2019-06-06:
I believe the lidar returns what it was able to scan in one frame based on the lidar's rotation speed, and then continues the scan in the next frame. You won't get a full lidar sweep every frame. So you ought to check whether you are actually getting all your data, even if it's split up across a few successive frames.

JeremyBYU, 2019-06-06:
> @JeremyBYU <https://github.com/JeremyBYU> I believe the lidar returns

Yes that what it seems to me as well. I was just surprised because it didnt
seem to update the final output (the output variable that the rpc server
returns) until after a scan has been complete . However it does call this
setoutput method as a reference so maybe that will makes the timing a moot
point.

In any case thank you for taking the time too look into this. Usually when
reading a lidar sensor you will have the a ability to gather points and
only "send" a message that might have a complete scan. It seems that this
would have to be manually handled on the client side after an rpc call.

Thanks!

closed 2019-06-06 by JeremyBYU

EivindYN, 2020-07-03:
AirSim tries to send something close to the whole scan, but I think there are a bug which causes the publishPointCloud to be called twice. Which makes it send half & half of a scan, or almost full & almost nothing of a scan, etc.

I was able to fix it so that it only sends the full scan by going into the UnrealLidarSensor.cpp inside the environment (changing any other UnrealLidarSensor.cpp in other locations wont have any affect). And adding "delta_time = 0.1f;" to the first line of the getPointCloud() function. Note: 0.1f is calculated by 1/rotationsPerSeconds, so you may need to use another number.

This forces the function to believe it must do a full scan in every cycle.