microsoft/AirSim #1998: Why? vechile can not MoveToPosition or MoveOnPath?

Hezzzzzy, 2019-06-04:
@madratman 
HI folks,
I try to control the vehicle by a high level way, I have read the python code, I find there are some discard api,  "movetoposition" & "moveonpath" in the vehicleclient, 

now, we can only use these api by control drone, 

can we use it now? it's very useful

msb336, 2019-06-17:
@Hezzzzzy 

Both of those APIs are still active, and you can use them in the python client.

closed 2019-06-17 by msb336

Hezzzzzy, 2019-06-18:
@msb336 hey bro, maybe you misunderstand me....I try to call python api "moveOnPath" on car, it returns me "moveOnPathAPI is deprecated. Please use moveOnPathAsync() API.",  then  I found this api only can use on drone, so I ask.....

and I searched this api in C++file, and it definited in MultirotorRpcLibClient.cpp, not in CarRpcLibClient.cpp, 

I worked in a project about High precision map， if moveOnPath api  could be used in car, it will be very awsome, so....