microsoft/AirSim #1999: No lidar with name Lidar1 exist on vehicle on running car_lidar.py

Kzernobog, 2019-06-05:
I realise this issue has already been dealt with in other issues. https://github.com/microsoft/AirSim/issues/1518#issue-377823789. I followed the steps mentioned in these issues. I made a fresh clone of AirSim, built it, copied the fresh plugins, and replaced the old plugin folder with this new one in my Unreal project. However, I seem to be facing the same problem even after performing the necessary steps as mentioned in these issues.

The background is that, I started working with a computer that already had an installation of AirSim, albeit an older one. I created a new Unreal project, included the plugins folder. Created my custom simulator and ran the application. I then ran the `car_lidar.py` script. I ran into the following error

`Connected!
Client Ver:1 (Min Req: 1), Server Ver:1 (Min Req: 1)

Go Forward
Go Forward, steer right
Press any key to get Lidar readings
Press any key to reset to original state
Done!

Traceback (most recent call last):
  File "car_lidar.py", line 93, in <module>
    lidarTest.execute()
  File "car_lidar.py", line 50, in execute
    lidarData = self.client.getLidarData(lidar_name='Lidar1')
  File "C:\Users\ash\Anaconda3\lib\site-packages\airsim\client.py", line 160, in getLidarData
    return LidarData.from_msgpack(self.client.call('getLidarData', lidar_name, vehicle_name))
  File "C:\Users\ash\Anaconda3\lib\site-packages\msgpackrpc\session.py", line 41, in call
    return self.send_request(method, args).get()
  File "C:\Users\ash\Anaconda3\lib\site-packages\msgpackrpc\future.py", line 45, in get
    raise error.RPCError(self._error)
msgpackrpc.error.RPCError: rpclib: function 'getLidarData' (called with 2 arg(s)) threw an exception. The exception contained this information: No lidar with name Lidar1 exist on vehicle.`

Here are the steps I followed
1. I cloned the recent master branch of AirSim
2. Built it again
3. copied the new plugins folder and replaced the old plugins folder in my Unreal project
4. generated new VS 2017 files, built it.

After launching the Unreal project and running `car_lidar.py` again, I face the same issue. The error is the same
`Connected!
Client Ver:1 (Min Req: 1), Server Ver:1 (Min Req: 1)

Go Forward
Go Forward, steer right
Press any key to get Lidar readings
Press any key to reset to original state
Done!

Traceback (most recent call last):
  File "car_lidar.py", line 93, in <module>
    lidarTest.execute()
  File "car_lidar.py", line 50, in execute
    lidarData = self.client.getLidarData(lidar_name='Lidar1')
  File "C:\Users\ash\Anaconda3\lib\site-packages\airsim\client.py", line 160, in getLidarData
    return LidarData.from_msgpack(self.client.call('getLidarData', lidar_name, vehicle_name))
  File "C:\Users\ash\Anaconda3\lib\site-packages\msgpackrpc\session.py", line 41, in call
    return self.send_request(method, args).get()
  File "C:\Users\ash\Anaconda3\lib\site-packages\msgpackrpc\future.py", line 45, in get
    raise error.RPCError(self._error)
msgpackrpc.error.RPCError: rpclib: function 'getLidarData' (called with 2 arg(s)) threw an exception. The exception contained this information: No lidar with name Lidar1 exist on vehicle.`

Im using Python 3.7.3, Unreal 4.18.

Thanks

msb336, 2019-06-05:
@Kzernobog you need to [specify which vehicle the sensor is applied to](https://github.com/microsoft/AirSim/blob/master/docs/sensors.md#configuring-vehicle-specific-sensor-list), or [specify that it is inside of the default sensor group](https://github.com/microsoft/AirSim/blob/master/docs/sensors.md#configuring-the-default-sensor-list). If sensor configuration settings are not inside of a Sensors catgetory, the json reader does not know how to interpret that data.

closed 2020-02-14 by ironclownfish

EdwinMeriaux, 2020-06-26:
> @Kzernobog you need to [specify which vehicle the sensor is applied to](https://github.com/microsoft/AirSim/blob/master/docs/sensors.md#configuring-vehicle-specific-sensor-list), or [specify that it [...]

Hey I'm getting the same error as @Kzernobog 
But I thought I specified the senesor to viehicle setup:

{
	"SettingsVersion": 1.2,
	"SimMode": "Multirotor",

	"Vehicles": {
		"Drone1": {
			"DefaultSensors": {
			    "Barometer": {
			         "SensorType": 1,
			         "Enabled" : true
			    },
			    "Imu": {
			         "SensorType": 2,
			         "Enabled" : true
			    },
			    "Gps": {
			         "SensorType": 3,
			         "Enabled" : true
			    },
			    "Magnetometer": {
			         "SensorType": 4,
			         "Enabled" : true
			    },
			    "Distance": {
			         "SensorType": 5,
			         "Enabled" : true
			    },
			    "Lidar2": {
			         "SensorType": 6,
			         "Enabled" : true,
			         "NumberOfChannels": 4,
			         "PointsPerSecond": 10000
			    }
			},
			"VehicleType": "SimpleFlight",
			"X": 120, "Y": -100, "Z": -2
		}
	}
}

That is my json file yet I am still getting:

Traceback (most recent call last):
  File "drone_lidar.py", line 93, in <module>
    lidarTest.execute()
  File "drone_lidar.py", line 47, in execute
    lidarData = self.client.getLidarData("","");
  File "/home/edwin-pcp/AirSim/PythonClient/airsim/client.py", line 537, in getLidarData
    return LidarData.from_msgpack(self.client.call('getLidarData', lidar_name, vehicle_name))
  File "/home/edwin-pcp/anaconda3/lib/python3.7/site-packages/msgpackrpc/session.py", line 41, in call
    return self.send_request(method, args).get()
  File "/home/edwin-pcp/anaconda3/lib/python3.7/site-packages/msgpackrpc/future.py", line 45, in get
    raise error.RPCError(self._error)
msgpackrpc.error.RPCError: rpclib: function 'getLidarData' (called with 2 arg(s)) threw an exception. The exception contained this information: No lidar with name Lidar2 exist on vehicle.