microsoft/AirSim #2003: Segmentation non existant on spawning of new actors

machampagne1, 2019-06-06:
Hi, I'm working on a project where every few second I delete all actors to spawn new ones (what for randomness) but whenever they respawn, none of the items have segmentation !

AFAIK AirSim only gives segmentation on the start of a project, how do I make it so it's continuous ? (at every spawn, give them segmentation)

Thanks !

msb336, 2019-06-06:
@machampagne1 I think you will need to set the stencil ID of your static mesh components in the spawned actors in order to see them in segmentation view. You can do this at creation of your Actors, or afterwards with the command `simSetSegmentationObjectID`

machampagne1, 2019-06-06:
Sorry, I'm kinda new to all of this (both AirSim & UE4)
These items respawn every second, I don't know if I can type this fast !

Is there a way to do it at the creation of the item in UE4 ? (like, with a blueprint command or something ?)

Or perhaps can I just create an infinite loop that just spams the command in the prompt ?

Thanks for your time !

msb336, 2019-06-06:
I was under the impression that you were controlling the item spawning. If that is the case, then yes, you can set stencil ID at creation.

Once you have isolated the UStaticMeshComponent of your actor, you'll need to run the following commands at creation of your actor:
```
your_mesh_component->SetCustomDepthStencilValue(desired_stencil_ID);
your_mesh_component->SetRenderCustomDepth(true);
```

machampagne1, 2019-06-06:
I do not control it, it's a blueprint within a blueprint (first one creates the actor that generates the items, the items automatically destroy themselves after a given ammount of time.)

idk how to control their spawning with cmd. but I can look it up and try that.

Thanks again !

msb336, 2019-06-06:
This is not something you should be doing in the command window. Modify your blueprints so they are setting a [CustomDepthStencilValue](https://api.unrealengine.com/INT/BlueprintAPI/Rendering/SetCustomDepthStencilValue/index.html) to the actors they are spawning.

machampagne1, 2019-06-06:
Sorta like this ?
[image]

I've tried putting the other right after but it doesn't seem to work.
This is in the child actor that creates the individual parts of the object

(yeah ik, it's a bit of a rabbit hole.. my item spawns an item that spawns a random assortment of meshes)

machampagne1, 2019-06-06:
Nevermind !
Thank you SO MUCH !!!!

It works !!!!!
It's a bit slow.. (though I am spawning like 1700 meshes at once, every second.. I think I can turn that down xD)

closed 2019-06-06 by msb336